mdhiggins/ESP8266-HTTP-IR-Blaster #92: Hostname not updating

juliusedward, 2017-12-07:
Using a Wemos D1 mini with D5 pin as config pin.
After configuration, hostname is updated but with a normal reset of the wemos board, the old hostname is set. Even if the json file is set with hostname: irblaster, the hostname is set to ESP_424167

Config pin GPIO5 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"**irblaster**","passcode":"********","port_str":"80","user_id":"amzn1.account.****************************","ip":"192.168.1.115","gw":"192.168.1.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.115
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.115
WiFi connected! User chose hostname **'ESP_424167'** passcode '********' and port '80'
mDNS started. Hostname is set to **ESP_424167.local**
Local IP: 192.168.1.115
URL to send commands: http://ESP_424167.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
External IP address request took 8987 ms
Ready to send and receive IR signals

juliusedward, 2017-12-07:
Complete console output

Entered config mode
192.168.4.1
IRBlaster Configuration
*WM: 
*WM: Configuring access point... 
*WM: IRBlaster Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Scan done
*WM: Julius_Extended
*WM: -55
*WM: PU
*WM: -68
*WM: Julius
*WM: -74
*WM: Bruno2.4
*WM: -74
*WM: Nitesh
*WM: -81
*WM: chandra1204
*WM: -86
*WM: Kishor
*WM: -97
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: irblaster
*WM: Parameter
*WM: passcode
*WM: ********
*WM: Parameter
*WM: port_str
*WM: 80
*WM: Parameter
*WM: user_id
*WM: amzn1.account.****
*WM: static ip
*WM: 192.168.1.115
*WM: static gateway
*WM: 192.168.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.115
*WM: Connection result: 
*WM: 3
Should save config
WiFi connected! User chose hostname 'irblaster' passcode '********' and port '80'
 config...
{"hostname":"irblaster","passcode":"********","port_str":"80","user_id":"amzn1.account.****","ip":"192.168.1.115","gw":"192.168.1.1","sn":"255.255.255.0"}
mDNS started. Hostname is set to irblaster.local
Local IP: 192.168.1.115
URL to send commands: http://irblaster.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
External IP address request took 8996 ms
Ready to send and receive IR signals

After a normal reset

ESP8266 IRBlaster
Config pin GPIO5 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"irblaster","passcode":"********","port_str":"80","user_id":"amzn1.account.****","ip":"192.168.1.115","gw":"192.168.1.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.115
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.115
WiFi connected! User chose hostname 'ESP_424167' passcode '********' and port '80'
mDNS started. Hostname is set to ESP_424167.local
Local IP: 192.168.1.115
URL to send commands: http://ESP_424167.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
External IP address request took 8987 ms
Ready to send and receive IR signals

mdhiggins, 2017-12-07:
Looks like this was related to the recent pull request, didn't notice that one line change
Should be fixed now, please update and give it a shot

juliusedward, 2017-12-08:
Works well with the new update. Thanks

closed 2017-12-08 by mdhiggins