mdhiggins/ESP8266-HTTP-IR-Blaster #95: Controller Device Setup Login Trouble

DataBasye, 2017-12-13:
Yesterday, when I went to https://tehpsyc.pythonanywhere.com/ and clicked the login button, it took me to a popup that only asked for my password.  Today, when I attempt the same, it takes me to a popup that asks for both my ID and password, and tries to log me into amazon itself. It gives me an error saying "The Web address you entered is not a functioning page on our site."

Did something change?

mdhiggins, 2017-12-13:
Should always ask for username and password and its working fine when I test it now

Give it a shot

DataBasye, 2017-12-15:
Thanks for quick response. Gave it another shot, but still having same issue.  Traced it to a Windows 10 update (#1607).  That's the only thing that has changed. Everything worked fine before that update.  Now when I try to log into the pythonanywhere site with my Amazon ID and password, it comes back with the following error dialog: We're sorry. The Web address you entered is not a functioning page on our site. Go to Amazon.com's Home Page.  It doesn't take me to your site. 

If you are using a mac, it is probably continuing to work fine because you won't have the OS update issue getting in your way. 

When I go to the pythonanywhere site, and I click on the "login with Amazon" button, I'm pretty sure the next screen where you enter the ID and password looks different than it did before the OS update. But I obviously can't get before and after pictures. 

Microsoft won't let me reverse out the update or restore back to the prior version of the OS.  So I'm stuck.

I don't know how you are working that part of the project. Does this mean I need to learn how to build an Alexa skill to get this piece working?

mdhiggins, 2017-12-15:
I use Windows most of the time and haven’t had any issues with the latest update. What browser?

Working fine on chrome for me

DataBasye, 2017-12-15:
Chrome is what I'm using too (Version 63.0.3239.84). I also tried Edge and IE. Same results. So it's not the browser itself.

I'm running Windows 10 build 16299.  That cumulative update is the only thing I can think of that changed.  

I'm stumped and it's driving me nuts!

mdhiggins, 2017-12-15:
Is it opening with https or http?

DataBasye, 2017-12-16:
All screens show https

mdhiggins, 2017-12-16:
Hm I have no idea honestly I can’t reproduce this. If you try from a phone or another device is it fine?

DataBasye, 2017-12-16:
Tried from another device and got it working.  

I was able to set up a device (a TV) and have Alexa discover it. When I ask Alexa to turn on the TV, she responds that the TV appears to be malfunctioning. I think it's because I don't have the right external IP address. 

When I go to the IR Controller local web address, you say in the tutorial that the external IP address should show at the top of the screen. I see where it is supposed to show, but what's there is just 0.0.0.0 followed by the port. Is there a step I missed? I'm not sure how to discover the external IP address of the ESP8266.  

Thanks for your help.

mdhiggins, 2017-12-16:
The external IP address is whatever your WAN IP is for your network
You can find this by accessing whatismyip.com from any device on the same network

That being said, if the device is unable to display or determine your external IP, this is usually because you don't have your local IP address settings (usually gateway or DNS) set appropriately and you will run into other issues further down

Also don't forget to forward the appropriate port to the device

closed 2017-12-17 by DataBasye