mdhiggins/ESP8266-HTTP-IR-Blaster #96: WiFi Manager custom entries

DataBasye, 2017-12-17:
I'm entering a custom host name in the WiFi manager screen.  When the device reboots into the IR controller mode, my custom host name gets clobbered with "ESP_4F1F01".  It doesn't recognize that I changed to a custom name. 

Similarly, when I enter my custom gw on the WiFi manager screen, it does not get carried over to the IR Controller screen for display at the top, or inserted in the generated URL message codes. As long as I manually insert the correct gw in the URL message code that I enter in the device setup, Alexa should work. Right?

mdhiggins, 2017-12-17:
When was the last time you updated? There was a bug for a few days with the hostname that was fixed on 12/7/2017 d46b429fbf0dc65e6f21c9263ad1bf7e148eddac so that issue should be resolved

As for the gateway, what do you think a gateway is? The gateway setting is literally just your default gateway for configuring your local IP address setting. It shouldn't appear at the top of the page and shouldn't appear in the URL

Its the equivalent of this setting in your IP address configuration

[image]

DataBasye, 2017-12-18:
Sorry about the gateway comment...I meant the external IP.

mdhiggins, 2017-12-18:
The default setting for external IP got set to disabled in a pull request that I missed. Just set that to true in the blueprint or update to the latest version where I reverted back to enabled as the default setting.

DataBasye, 2017-12-18:
Okay, thanks.  

By the way...the host name is still not getting saved properly. 

When I went into Wifi manager and changed the name the first time, it worked.  However, I had a space in the name that I wanted to get rid of. When I went back in a second time, it didn't take my updated custom name, and reverted back to ESP_4F1F01.  In fact, it reset all my fields to original, so that I had to retype everything rather than just alter the hostname. I'm working around it, but there's still a bug in there somewhere. 

Thanks for your help on this great project!  I've got Alexa talking to my ESP now.

mdhiggins, 2017-12-18:
I made some more changes to the way host_name is handled if you wanna update and try again
I tested it pretty thoroughly and can't seem to recreate the error you were having before

Let me know

DataBasye, 2017-12-20:
When I downloaded the latest version, I'm getting a compile error: 'ArduinoJson::DynamicJsonBuffer' has no member named 'clear'.  I thought I had the latest versions of all the libraries.  Is there a chance the 'clear' function got deleted in a newer library version?

mdhiggins, 2017-12-21:
Not able to reproduce this
Compiling fine on ArduinoJson version 5.12.0 which appears to be the latest
Looks like on the latest build on their github clear is still there
What version you compiling against?

DataBasye, 2017-12-28:
My fault.  Had multiple library copies and the wrong one was getting picked up at compile time. Sorry.

closed 2017-12-28 by DataBasye