mdhiggins/ESP8266-HTTP-IR-Blaster #103: No file found .....#include "sha256.h"

jainhemendrah, 2017-12-21:
Dear,
  I downloaded the latest IR Controller code. and followed the steps to download the other libraries.
It seems the local file  "sha256.h" is missing

mdhiggins, 2017-12-21:
You didn’t install Cryptosuite

closed 2017-12-21 by mdhiggins

jainhemendrah, 2017-12-21:
Hi ,
  I installed CryptoSuite and then I kept getting error io/avr.h not found.
This was reported to be from sha256.cpp file

mdhiggins, 2017-12-21:
What version of cryptosuite did you install? Can you link me

mdhiggins, 2017-12-21:
Also if you’re on Linux you make just be missing some critical libraries

Install these at the operating system level

gcc
gcc-avr
avr-libc
avrprog
avrdude

jainhemendrah, 2017-12-21:
I installed the latest CrytoSuite library.

Sure, Will install gcc,gcc-avr,etc...thanks Michael

jainhemendrah, 2017-12-21:
I installed  gcc,gcc-avr,.. files.
Again, opened IR  Controller and tried compiling.

Got following errors:

\Sha\sha256.cpp:2:20: fatal error: avr/io.h: No such file or directory #include <avr/io.h>

mdhiggins, 2017-12-21:
I would post on cryptosuite GitHub and ask the. I’m not sure

jainhemendrah, 2017-12-21:
You are right. The issue is with CryptoSuite....