mdhiggins/ESP8266-HTTP-IR-Blaster #105: Mess about address filed according to panasonic codes

rselec, 2017-12-27:
Hi @ All,

we have a huge mess around the address filed used by panasonic codes.

When a code was scanned the address was added to the command in the data-field and also in the suggested links. But when sending a panasonic code a seperate address tag is requiered and the address have to be removed from the command!
Also you have to type 0x4004 instead of 4004 because of the c_str format detection, but the command is converted by a special conversion function, so no 0x is needed. Thats not consistance.

I have some fixes in my fork https://github.com/rselec/IR-Gateway but I haven't cleand up the whole mess.

So any suggestions how we're going to hande the address filed in future? As seperate tag or as part of the command?

mdhiggins, 2017-12-28:
I'm not sure specifically why this is a mess

Its working as intended and the usage of strtoul format was a nice solution that allows people to specify the address as DEC or HEX and have it get converted to the right long value

And the handling is already done. Its a special parameter only need by certain formats and can be specified when those formats are used

I'm not clear what your specific concerns are

rselec, 2017-12-28:
The generated link shown on the webintreface won't work, if you use the panasonic protocol!

When a Panasonic code is received the generated URLs doesn't contain the neccessary address field (see lines 1054 to 1069) and selCode.data has the address attached in the first 16 bits.

But to send the code you need the code without address (like in selCode.command) and the seperate address field (compare lines starting at 692).

That is annoying and you have to dig into the code to figure out what to send.

Best, Robert

mdhiggins, 2017-12-28:
Alright I think I understand the 2 primary complaints you are raising

1. You don't like that there isn't a shorthand notation that includes the address parameter, (current shorthand is `data:type:length`

2. The example URL for panasonic commands doesn't include the address parameter

Number 2 is completely valid and I just pushed a commit that will include the address the URL for PANASONIC type commands
1 is a personal preference and I don't see any reason to change it at this point. Panasonic is the only encoded command where address matters at this point so adjusting the shorthand for the one type seems odd. The shorthand notation is kind of a relic and left in backwards compatibility situation from version 1 of the script that still used BASIC, not something I'd want to change.

This only applies to the older `/msg` URL formatting and not to the JSON style formatting anyway

So in summary gonna keep address a separate option to preserve the `/msg` formatting which only really exists as a backwards comparability measure anyway but I have updated the example URLs to include address for Panasonic in both formats

mdhiggins, 2017-12-28:
2df392b3bc9ab7cbf002284c9d66e4626597b7a0

closed 2017-12-28 by mdhiggins