mdhiggins/ESP8266-HTTP-IR-Blaster #107: More compatibility

glabbe0299, 2017-12-27:
Hi I managed to make it work so the Led is transmitting but however it doesn't seem to learn the code right as the code sent does nothing to my appliances.
My Tv is an LG and my receiver a Pioneer. Is there something I'm doing wrong ?

mdhiggins, 2017-12-28:
Need significantly more information

What codes are you trying to send?  How are you confirming that the LED is transmitting? What resistors are you using? How far away?

glabbe0299, 2017-12-28:
I’m Learning codes to turn off and on my Pioneer Amp and Lg tv. The led is transmitting as for the resistors I’m kind of a newbie and don’t really know what ohm to use. The leds are from an Ir blaster extension that came with a kit from a computer. I’m trying to send IR from 6 inch away or 2 feet no difference. Ir is coming out but it seems faster than the actual remote when seeing it from a phone camera.

Thanks for the help.

Provenance : Courrier pour Windows 10

De : Michael Higgins
Envoyé le :27 décembre 2017 22:47
À : mdhiggins/ESP8266-HTTP-IR-Blaster
Cc : glabbe0299; Author
Objet :Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] More compatibility (#107)

Need significantly more information
What codes are you trying to send? How are you confirming that the LED is transmitting? What resistors are you using? How far away?

mdhiggins, 2017-12-28:
What commands are you sending?

glabbe0299, 2017-12-28:
I use the commands received by the software when I press the remote buttons aimed at the receiver. The module throws multiple results when I press on one button I don’t know if it’s normal.
As you can see when I press the power button for the amp at 23 :55 :25 I get 2 codes. Pressed it a second time and now one code 1 second later. 

Provenance : Courrier pour Windows 10

De : Michael Higgins
Envoyé le :27 décembre 2017 22:52
À : mdhiggins/ESP8266-HTTP-IR-Blaster
Cc : glabbe0299; Author
Objet :Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] More compatibility (#107)

What commands are you sending?

mdhiggins, 2017-12-28:
Can you post the codes and the URLs you're using to send them (you can strip out your IP address and pass)

And if you're getting 2 codes detected its probably a bad detection. There's noise and you often have to scan the code a couple times to get the right one

LG TVs usually use NEC encoding, not sure what Pioneer uses

mdhiggins, 2017-12-28:
In fact if its a newer LG there are hidden discrete on/off codes that don't toggle that are nice to use that I happen to know off hand
On: 20DF23DC:NEC:32
Off: 20DFA35C:NEC:32

glabbe0299, 2017-12-28:
http://xxx.xxx.xxx.xxx:xx/json?plain=[{'data':'A55AD02F','type':'NEC','length':32},{'data':'20DF10EF','type':'NEC','length':32},{'data':'FFFFFFFFFFFFFFFF','type':'NEC','length':0}]

Provenance : Courrier pour Windows 10

De : Michael Higgins
Envoyé le :27 décembre 2017 23:00
À : mdhiggins/ESP8266-HTTP-IR-Blaster
Cc : glabbe0299; Author
Objet :Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] More compatibility (#107)

Can you post the codes and the URLs you're using to send them (you can strip out your IP address and pass)
And if you're getting 2 codes detected its probably a bad detection. There's noise and you often have to scan the code a couple times to get the right one
LG TVs usually use NEC encoding, not sure what Pioneer uses

mdhiggins, 2017-12-28:
The FFFFF command is basically a failed read command, you can strip that out as it won't do anything

The other 2 though look appropriate and are at least validly formatted codes and should work, so then I guess the question is do you have things wired up correctly. What ohm resistors are you using? and if you want, you can post a picture of your hardware and I can double check the wiring

glabbe0299, 2017-12-28:
I only have a 1K resistor from the 2n2222 to the board I did not have low enough resistor to put between the led and the board.
Everything fires up when asked I think I just have to experiment a bit more with Learning the codes. 

Provenance : Courrier pour Windows 10

De : Michael Higgins
Envoyé le :27 décembre 2017 23:04
À : mdhiggins/ESP8266-HTTP-IR-Blaster
Cc : glabbe0299; Author
Objet :Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] More compatibility (#107)

The FFFFF command is basically a failed read command, you can strip that out as it won't do anything
The other 2 though look appropriate and are at least validly formatted codes and should work, so then I guess the question is do you have things wired up correctly. What ohm resistors are you using? and if you want, you can post a picture of your hardware and I can double check the wiring

mdhiggins, 2017-12-28:
I would first try and get the codes to work individually then try grouped together

http://xxx.xxx.xxx.xxx:xx/msg?code=A55AD02F:NEC:32 for example and once you've confirmed that works then expand to the combo of the two

glabbe0299, 2017-12-28:
I will try to do that and confirm what I found.

Thank you 

Provenance : Courrier pour Windows 10

De : Michael Higgins
Envoyé le :27 décembre 2017 23:09
À : mdhiggins/ESP8266-HTTP-IR-Blaster
Cc : glabbe0299; Author
Objet :Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] More compatibility (#107)

I would first try and get the codes to work individually then try grouped together
http://xxx.xxx.xxx.xxx:xx/msg?code=A55AD02F:NEC:32 for example and once you've confirmed that works then expand to the combo of the two

mdhiggins, 2017-12-28:
If they work individually but not together you can try adding a delay between the two and see if its just trying to do them both too fast

kaotix, 2017-12-29:
I thought I'd jump in here. Having a similar issue as I also have an LG TV where I can get it to turn ON but not OFF.
I tried the codes mdhiggins suggested for OFF but that didn't work.
The code I get for ON/OFF button is 20DF10EF:NEC:32
My TV is LG 49UH603V

How much difference do the resistors and transistors have on the effectiveness of the IR LEDs?
I currently have my setup sat right in front of the TV and ON works flawlessly every single time.

glabbe0299, 2017-12-29:
I managed to make the Tv work when the IR led is 6 inch from the tv but the pioneer amp gives me trouble.
When I scan the Remote code with the ESP receiver i get 3 different raw codes. Not sure what to do with it.
I activated the HDMI control on the amp so when the tv goes off the amp shuts down too. I just hav to powerup the amp manually for the time.

mdhiggins, 2017-12-29:
@kaotix resistors can make a huge difference in range
@glabbe0299 the pioneer signal is a RAW command or 3 RAW commands pop up with each button press? Feel free to post those codes here; also what Pioneer receiver is it?

glabbe0299, 2017-12-29:
[image]
This is what it looks like when I press the power button one time.
It's a Pioneer VSX-921 
I found the Hex code to power it up but can't manage to interchange with nec commands

mdhiggins, 2017-12-29:
That NEC command doesn't work to power up the device? It looks like it should
Because it gets sent at least twice, you could try adding a `pulse` parameter to your command and see if that makes it turn on, start with 2 and go up and see if it starts responding

mdhiggins, 2017-12-29:
Per the pioneer website these commands should work for that model:

Power On: `A55A58A7:NEC:32`
Power Off: `A55AD827:NEC:32`
Power Toggle: `A55A38C7:NEC:32` (consistent with the one you scanned)

glabbe0299, 2018-01-04:
I didn't had the time to try it and I will probably need an other Ir led to control the Amp as it is too far from the tv ir so I will probably test it this weekend. Thanks for the support.

jskubick, 2018-01-04:
Try explicitly specifying "khz" = "33". Pioneer has historically used NEC protocol, but with an oddball 33KHz carrier frequency.

mdhiggins, 2018-01-06:
You may need to send the signal as RAW if you're going to adjust the kHz though

mdhiggins, 2018-01-12:
Closing due to inactivity

closed 2018-01-12 by mdhiggins