mdhiggins/ESP8266-HTTP-IR-Blaster #108: NodeMCU rebooting after a few seconds - continuous loop

kaotix, 2017-12-28:
I have just flashed this to my nodemcu but when it starts running i get the 'ESP8266 IR controller' output in console then the device reboots and just keeps doing this over and over.
I have all default settings at the moment just to test. 
NodeMCU 0.9 (esp-12 module), 80mhz, 115200, 4m (3m spiffs)

` ets Jan  8 2013,rst cause:4, boot mode:(3,2)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
)
`

mdhiggins, 2017-12-28:
wdt reset means its crashing

Why exactly is usually tough to pinpoint, but here's some things to start with
Make sure you're using the latest version of the code
If you have an extra nodemcu board, try that and see if its reproducible, these do experience hardware failures from time to time
Disconnect everything from the nodemcu (remove from breadboard) and try running it isolated and see if it boots

If those steps don't work let me know

billcramer07, 2017-12-29:
I had the exact same problem using a Adafruit Huzzah. Found out it did not like the wifi reset on GPIO 10. Make line 35 of the code = to 12. Even  though 12 is already defined as IR send pin 3 it should boot into AP mode. Also u must change line 46 to the static IP u want and change line 47 to your gateway address. Even though these can be changed in setup it won't take them.

mdhiggins, 2017-12-29:
While I agree with the first statement @billcramer07 made, there's no reason overriding the WiFiManager settings should have any impact on booting

billcramer07, 2017-12-29:
Everytime I would run WiFiManager and enter my gateway and static IP, WiFiManager would override it with the defaults at lines 46 and 47.

billcramer07, 2017-12-29:
BTW, your video doesn't show the 3 additional lines in Setup to enter the IP, gateway and subnet mask. Took a while to figure it all out. EXCELLENT SKILL FOR ALEXA!!!

mdhiggins, 2017-12-29:
Those are just placeholder values that are overwritten before they are used line 334-336 from the values stored in the config.json file from WiFiManager
The values are assigned to the staticIPConfig on line 360

You have to have a placeholder value set so that when WiFiManager enters config mode it will see that it’s a static configuration and prompt for the values for the IP/GW/SN

From: billcramer07
Sent: Thursday, December 28, 2017 10:59 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] NodeMCU rebooting after a fewseconds - continuous loop (#108)

Everytime I would run WiFiManager and enter my gateway and static IP, WiFiManager would override it with the defaults at lines 46 and 47.

mdhiggins, 2017-12-29:
Yeah the video was made when everything was dynamic, I added the static stuff after the fact to make port forwarding more stable

From: billcramer07
Sent: Thursday, December 28, 2017 11:05 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] NodeMCU rebooting after a fewseconds - continuous loop (#108)

BTW, your video doesn't show the 3 additional lines in Setup to enter the IP, gateway and subnet mask. Took a while to figure it all out. EXCELLENT SKILL FOR ALEXA!!!

kaotix, 2017-12-29:
I managed to get it working on an ESP-M3. I will have a look.
I can't remember if I tried changing the configpin on these other devices, I'll give that a go later though.

So far I've captured some codes for my TV and they appear to be working with Alexa. It's actually really simple once you get it all up and running which is brilliant.

mdhiggins, 2018-01-04:
Gonna close this for now, reopen if you're still having issues

closed 2018-01-04 by mdhiggins