mdhiggins/ESP8266-HTTP-IR-Blaster #110: Config file and WiFi Manager

DataBasye, 2017-12-30:
Mike, 
Great project, and I have had partial success.

I re-downloaded a fresh copy of the blueprint a couple days ago to make sure I picked up any resent changes.  I'm still getting some inconsistent results customizing config settings through WiFi Manager.  

I have gotten to the point where Alexa can turn on/off my TV, but I can no longer access the server page that shows me the codes it is sending/receiving. I could before. I've tried accessing via the IP address and via the NAME.local:port address. Neither are working for me.  I get the message: "This site can't be reached. Server IP address could not be found." Yet I know the server is logged into my network. I can see it is active, and I can still successfully send commands via Alexa. It still turns the TV on/off.

I would appreciate any suggestions you might have for what I should be checking. 

It seems like sometimes my custom WiFi Manager settings get saved and used, and sometimes they don't. As I've been trying to pinpoint my problem, I have seen the Bonjour name differ from the DHCP name. I'm not familiar with Bonjour. Should these generally match? 

At one point, I saw the WiFi Manager routine throw an exception when trying to write the config file. There is no exception handling for this event, and I don't know what caused the exception.  The messages sent to the serial output were a little cryptic--"Exception #2" followed by some hex codes. Since the file was never written, the custom settings I entered in the WiFi Manager screen weren't saved or used.    

The WiFi Manager piece seems to add a lot of complexity to the project--both code and process.  Is there a reason you wanted the WiFi Manager process in the project instead of simply giving users a section in the blueprint to hardcode their customized configuration settings? I'm thinking of trying to create a simplified version of the blueprint that does this. I'm thinking the code would be smaller and simpler to use without jeopardizing any security.  Thoughts?

mdhiggins, 2017-12-30:
I suspect your issue with not being able to access the site has more to do with an issue with Bonjour than a problem with the device or WiFi manager. Can you access the page via the local IP address?

As for the config file, writing the config is not part of WiFi manager. The flash memory on the ESP is decent but will occasionally throw errors when being written, it’s just the nature of the file system it uses and the device not being as robust. If you encounter an except during writing just reinput the settings. It’s good and reliable when reading but you do occasionally need to try twice to get the settings written. But this is not part of WiFiManager, all that does is gather the input data and forward it to the JSON config writer. 

The reason for WiFi manager is that it allows the device to be setup and administered without needing to modify the blueprint. This makes updating easier, is more user friendly, and doesn’t store dynamic variables as hard coded options in the blueprint. Hope that explains things.

DataBasye, 2017-12-30:
No, I can't access the page via the local IP either.  I get the message "site can't be reached...refused to connect." The error message suggested that I check the proxy and firewall.  I guess I could look at those, but I didn't have to change any of that before--especially since it is all local.  Maybe if I work out the writing of the config file first, the rest will fall into place.

Thanks for the additional explanations regarding the file writing and the reason for WiFi manager.

mdhiggins, 2017-12-30:
If the Alexa commands are working and the device is working remotely I would be highly suspicious something is going on with your local network that could be creating this issue
Perhaps the two devices aren’t on the same subnet or something but I wouldn’t go rewriting a bunch of code until you verify this. You can boot the device and look at the serial output to get all the IP address information, then verify that is on the same subnet as the computer you’re using to try to access the device. Also make sure you didn’t accidentally which the gateway and subnet and that the IP addresses have the same formatting for the first 3 digit sets

From: DataBasye
Sent: Saturday, December 30, 2017 12:23 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Config file and WiFi Manager(#110)

No, I can't access the page via the local IP either. I get the message "site can't be reached...refused to connect." The error message suggested that I check the proxy and firewall. I guess I could look at those, but I didn't have to change any of that before--especially since it is all local. Maybe if I work out the writing of the config file first, the rest will fall into place.
Thanks for the additional explanations regarding the file writing and the reason for WiFi manager.

closed 2018-01-08 by mdhiggins

DataBasye, 2018-01-12:
Sorry for delay.  I got tied up over the holidays. I checked the subnet and
made sure I wasn't switching the gw with the ip of the ESP device.  All
looked good there.  At some point, the device kept booting into WiFi
manager mode, and would never come out of that into regular mode.  I
decided to wipe the ESP8266 clean and try again.  I haven't had a chance to
get back to it since then.  If I discover anything, I'll let you know.
Thanks.