mdhiggins/ESP8266-HTTP-IR-Blaster #112: Alex Skill open source?

ljay79, 2018-01-01:
Hi.

Im new to this project but following it for almost a year now. Prepping for developing it according to your guides now.
A few small questions i couldnt find answer yet:
- is the Alexa Skill itself somewhere opensource available?
- do you have a schematic overview of the communications between all involved devices and services, im concerned of external (third-party) dependencies and .. yes, also my data privacy :)
- I am more into web development then micro controller solding and programming... so i might sound stupid: Is it possible to assemble an array of IR Transmitter to the ESP8266 in parallel so it would transmit same signal on all at same time. You describe a way of multiple IR LEDs to choose direction, but i would simply prefer sending everything in (all) directions so i wont have to care about the positioning of the device later that much...

In case these questions have been answered already (i couldnt find) pls be so kind an point me to the answers.
Thank a lot.

Great project by the way!

mdhiggins, 2018-01-04:
- Skill is not open source, might eventually release it open source one day when I get time
- No I'm afraid I do not, I'm not a professional developer or anything, don't even work in IT, just one guy who put this together for my friends and family then opted to share it. Its not particularly complicated though, Alexa queries the pythonanywhere server for a discover packet when it makes a discovery request. The discovery request contains all the URLs for the various commands as specially label 'cookies'. This is all done over HTTPS deliverying the JSON object visible at the bottom of the pythonanywhere page. Commands to the devices all happen on the AWS lambda servers which communicate directly to the devices, no pythonanywhere queries are made to turn your devices on/off for example. The only exception is channel changing where if you request a named channel instead of a numerical channel a second query is placed to the pythonanywhere server for the channel information. Data is stored in a sql database with your user_id and your JSON object that acts as the Alexa discovery object, and any custom channels you defined. All communication sent to the device can be authorized with a SHA256 HMAC encrypted time sensitive key to allow unauthorized communication to be filtered (valid for 30 second window). The SHA256 solution was implemented because the ESP8266 does not have enough memory to handle HTTPS validation (some fringe cases people have gotten it to work but it wouldn't be able to do all the other functionality) so the commands needed to be handled via HTTP
- You can wire them in series but you'll loose brightness, I suppose you could wire them in parallel too not sure which would be more ideal. Wiring in parallel you preserve voltage at the expensive or current, wiring in series you loose voltage and maintain current. It would depend on the operating voltage for your LED

Hope those answers help. Feel free to close if no further questions

ljay79, 2018-01-04:
Awesome thank you.
Initial questions answered, will close it. Thx a lot

closed 2018-01-04 by ljay79