mdhiggins/ESP8266-HTTP-IR-Blaster #114: Thank You!

benlye, 2018-01-02:
In the absence of any other way to give you feedback and/or show appreciation, I just wanted to say thanks for creating and sharing this!

I've been able to use your project get my Echo to control my TV, AV Receiver, and Sky Q set-top box, and it was pretty easy to accomplish.  There are a few rough edges (I'm still trying to figure out input names which will work and make sense), but I understand what the limiting factor is there.

You should put a Paypal donate button on here - I'd definitely send you a few bucks to buy you a beer 😄

paac, 2018-01-03:
Piling on to this, I've been planning to build an IR blaster for years now, and when I finally got around to putting a build together, I stumbled across one of your reddit submissions for this project of yours. I bought the parts off of AliExpress for less than $10, and after waiting several weeks for everything to show up, I was able to put together a working prototype in no time at all. Great work.

mdhiggins, 2018-01-04:
Glad you both are enjoying, no need for donations but I appreciate the sentiment
I put it together because I too was looking for an easy solution and it didn't look like one was out there

The input names are getting better, Amazon seems to be expanding the range of acceptable options. Its way better than when the API was in beta

closed 2018-01-04 by mdhiggins