mdhiggins/ESP8266-HTTP-IR-Blaster #116: strncpy (class Code.address[20] <> 40) size too big!

phili76, 2018-01-04:
just had an issue in my sketch when receiving ir data which overflows, the ESP hangs and restart. My first thoughts were regarding heap or stack so I reduced the size of some arrays. Then I found some strncpy which copy far more than they should. 😱 

````
class Code {
public:
    char encoding[14] = "";
    char address[20] = "";
    char command[40] = "";
    char data[40] = "";
    String raw = "";
    int bits = 0;
    char timestamp[40] = "";
    bool valid = false;
};
````

e.g.
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/7934290bf808ca229d1ba29b2bc7d4719a997711/IRController/IRController.ino#L770
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/7934290bf808ca229d1ba29b2bc7d4719a997711/IRController/IRController.ino#L1100
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/7934290bf808ca229d1ba29b2bc7d4719a997711/IRController/IRController.ino#L1111
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/7934290bf808ca229d1ba29b2bc7d4719a997711/IRController/IRController.ino#L1114
and:
1312
1351
1352
1398
1400

philipp

mdhiggins, 2018-01-06:
Good catch, these should all be fixed now

mdhiggins, 2018-01-06:
775c920e33895f9b85a54af99d94fc1653472134

closed 2018-01-06 by mdhiggins