mdhiggins/ESP8266-HTTP-IR-Blaster #117: Exception 28

nicomania, 2018-01-05:
Hello!
First of all, thank you for this great project. I tried to use it on a D1 mini board and only changed the Config Pin to GPIO2 (D4). But after configuring it over the AP, I get an exception message over the serial output:

```
Should save config
WiFi connected! User chose hostname 'xxxxxxxxx' passcode 'xxxx' and port '80'
 config...
{"hostname":"xxxxxx","passcode":"xxxx","port_str":"80","user_id":"xxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}

Exception (28):
epc1=0x4023b207 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000

ctx: cont 
sp: 3fff3790 end: 3fff3be0 offset: 01a0

>>>stack>>>
3fff3930:  0000001e 00000001 4010420e 3fff05c0  
3fff3940:  40235618 3fff4a44 3fff45b4 3fff6de8  
3fff3950:  3fff6cea 3fff5a9c 00000000 40220570  
3fff3960:  40103e13 3fff05c0 3fff4c38 3fff5a9c  
3fff3970:  3fff6cea 40104456 3fff4a44 3fff6de8  
3fff3980:  0000002e 3fff405c 3fff4a44 401043d1  
3fff3990:  3fff6ca4 3fff05c0 3ffefd18 00000000  
3fff39a0:  4010320d 00040000 7fffffff 3fff4091  
3fff39b0:  3fff405c 3fff405c 3fff6ca4 4010453d  
3fff39c0:  402275e7 3fff6ca4 3fff6cea 402275f0  
3fff39d0:  00000008 3fff3a00 50800000 4000050c  
3fff39e0:  3fffc278 40101f8c 3fffc200 3fff6cf8  
3fff39f0:  3fff6ca4 3fff6ef0 3fff405c 4022d681  
3fff3a00:  005e0001 0000fb00 0000000c ffffffff  
3fff3a10:  00000018 3fff6ef0 3fff6ca4 4022e108  
3fff3a20:  00000016 00000016 00000000 00000000  
3fff3a30:  3fff3aa4 00000001 00000000 00000002  
3fff3a40:  3fff3fa8 00000004 00000000 4022e778  
3fff3a50:  00000000 3ffeb5d8 00000000 3fff6eec  
3fff3a60:  00000000 3fff6ca4 3fff6d10 4022e148  
3fff3a70:  3fff405c 3fff3aa0 00000004 3fff6eec  
3fff3a80:  00000016 3fff405c 3fff6eec 4022d8d9  
3fff3a90:  3fff405c 3fff3aa0 00000004 4022764c  
3fff3aa0:  00000494 0104a8c0 3fff41b4 40227688  
3fff3ab0:  3fff6ef0 3fff405c 3fff5534 4022d9aa  
3fff3ac0:  3fff678c 3fff2644 3fff3b70 00000020  
3fff3ad0:  3fff3b3c 3fff405c 3fff6eec 4022db83  
3fff3ae0:  3fff3b3c 3ffebd8c 3fff405c 4022dc0f  
3fff3af0:  3fff5e0c 00000001 3fff3b70 00000001  
3fff3b00:  3fff2678 3fff2640 3fff2678 40219662  
3fff3b10:  00000000 3fff2640 3fff3b60 40100690  
3fff3b20:  3fff2ac4 0000063e 00000000 4021eb48  
3fff3b30:  00000000 3fff2640 3fff26a4 fb0000e0  
3fff3b40:  00000000 3fff2640 3fff2678 3fff2640  
3fff3b50:  00000000 3fff2640 3fff2678 4021ab2a  
3fff3b60:  3fff730c 3ffec254 40219354 40219784  
3fff3b70:  00000000 00000000 402071dc 40207632  
3fff3b80:  3fff2678 3fff2aec 3fff25e0 4020aafb  
3fff3b90:  feefeffe feefeffe feefeffe feefeffe  
3fff3ba0:  feefeffe feefeffe feefeffe feefeffe  
3fff3bb0:  feefeffe feefeffe feefeffe 3fff2bac  
3fff3bc0:  3fffdad0 00000000 3fff2ba4 4021ed74  
3fff3bd0:  feefeffe feefeffe 3fff2bc0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

ESP8266 IR Controller
Config pin GPIO2 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"xxxxxx","passcode":"xxxx","port_str":"80","user_id":"xxxxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.178.98
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.98
WiFi connected! User chose hostname 'xxxxxxxx' passcode 'xxxx' and port '80'
mDNS started. Hostname is set to ircontroller.local
Local IP: 192.168.178.98
URL to send commands: http://ircontroller.local:80
HTTP Server started on port 80
Error retrieving external IP
External IP address request took 5 ms
Ready to send and receive IR signals
Turning off the LED to save power.
Signal received:
One line: DCC8CD07:UNKNOWN:4
uint16_t  rawData[7] = {170,1044, 146,1070, 124,4538, 118};  // UNKNOWN DCC8CD07
```
any idea?

best regards, Nico

billcramer07, 2018-01-05:
Hi Nico,
I get the same message when I run wifi manager. Don't know why either, but it doesn't seem to affect operation. It will send commands from Alexa just fine and receive commands ok. BTW, hold the remote VERY close to the IR receiver when learning codes get a good read.
Bill C.

nicomania, 2018-01-06:
Hey Bill,
thanks for the quick answer! Yes you are right. After some resets, it started to connect and worked by receiving ir commands. Also the webinterface is reachable. But it still is not really usable. When I try to send a command 
http://ircontroller.local/msg?code=E0E040BF:SAMSUNG:32&pass=xxxx
I get the following error:

`
Connection received - MSG
Warning, user_id appears to be in the wrong format, security check will most likely fail. Should start with amzn1.account.***
Failed security check, signature is too old
Server: 0
Local: 143
MID: 
`

The AP configuration interface differs from the starting description here on github. I get asked for my amazon login (my email adress?), which obviously seems not to work. I also need to type in the ip adress, gateway and subnet mask manually. I would prefer to let my router deal with it, like it seems to work in the older versions.

As mentioned I changed the config pin to D4. But this also does not work. If tied to GND it does nothing. If i press reset additionally, I get strange symbols in the serial output forever. After a clean reset (without config pin) it just resets with old settings. Also reflashing via Arduino does not restore default settings.

Can you give me a hint please?

best regards, Nico

billcramer07, 2018-01-06:
Good morning, 
If you will go to the Alexa setup page https://tehpsyc.pythonanywhere.com/ and scroll to the bottom of the page and look at the very last line in small print, copy and paste this. It will solve your first issue. Also which esp board are you using. If it's a node MCU v.1 the reset pin should work as is.

nicomania, 2018-01-06:
Thank you again bill. I think I got the right id now. But this is still not the solution.
At "Force WiFi Reconfiguration" there is written "Set GPIO10 to ground". I managed to take the board to the AP mode by **pulling high** the configpin. Maybe this is a typo. I am using a Wemos D1 Mini board, which does not have the GPIO10 on the headers, so I changed it to GPIO2 (D4). 

I was still not able to set the ip to automatic, so I tried it with my manual settings as before. But it does not connect anymore. Please see the serial output below. After "Connection result: 0" it is switching to AP mode for configurtion. I also still get the exception 28. I did not change anything else, than the amazon id. It is so exausting and fragile......

```
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: ircontroller
*WM: Parameter
*WM: passcode
*WM: xxxx
*WM: Parameter
*WM: port_str
*WM: 80
*WM: Parameter
*WM: user_id
*WM: amzn1.account.AFWWR5AFP2X2BHDNxxxxxxxxxxx
*WM: static ip
*WM: 192.168.178.98
*WM: static gateway
*WM: 192.168.178.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.178.98
*WM: Connection result: 
*WM: 3
Should save config
WiFi connected! User chose hostname 'ircontroller' passcode 'xxxx' and port '80'
 config...
{"hostname":"ircontroller","passcode":"xxxx","port_str":"80","user_id":"amzn1.account.AFWWRxxxxxxxxxxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}

Exception (28):
epc1=0x4023b207 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000

ctx: cont 
sp: 3fff3790 end: 3fff3be0 offset: 01a0

>>>stack>>>
3fff3930:  000000b4 00000000 00000001 00000000  
3fff3940:  40235618 3fff48b4 3fff44f4 3fff6db8  
3fff3950:  3fff6e92 3fff579c 00000000 40220570  
3fff3960:  000005e0 00000020 401030c9 3fff05c0  
3fff3970:  3fff6e92 40104456 3fff48b4 3fff6db8  
3fff3980:  0000002e 3fff405c 3fff48b4 401043d1  
3fff3990:  3fff6e4c 00000566 00000000 3fff0230  
3fff39a0:  3ffefd18 3fff3abc 3fff6c4c 3fff4091  
3fff39b0:  3fff405c 3fff405c 3fff6e4c 4010453d  
3fff39c0:  402275e7 3fff6e4c 3fff6e92 402275f0  
3fff39d0:  00000008 3fff3a00 000005e3 4010020c  
3fff39e0:  40103d1d 3fff05c0 3ffefd18 3fff6ea0  
3fff39f0:  3fff6e4c 3fff5538 3fff405c 4022d681  
3fff3a00:  005e0001 0004fb00 7fffffff 00000000  
3fff3a10:  00000018 3fff5538 3fff6e4c 4022e108  
3fff3a20:  3fff05c0 40102136 50800000 00000000  
3fff3a30:  3fff3aa4 00000001 00000000 00000002  
3fff3a40:  3fff3fa8 00000004 3fffc200 4022e778  
3fff3a50:  3fff3a60 00000030 0000000c 3fff5534  
3fff3a60:  00000000 3fff6e4c 3fff6eb8 4022e148  
3fff3a70:  3fff405c 3fff3aa0 00000004 3fff5534  
3fff3a80:  00000016 3fff405c 3fff5534 4022d8d9  
3fff3a90:  3fff405c 3fff3aa0 00000004 4022764c  
3fff3aa0:  00000494 0104a8c0 3fff41b4 40227688  
3fff3ab0:  3fff5538 3fff405c 3fff70a4 4022d9aa  
3fff3ac0:  3fff7084 3fff2644 3fff3b70 00000020  
3fff3ad0:  3fff3b3c 3fff405c 3fff5534 4022db83  
3fff3ae0:  3fff3b3c 3ffebd8c 3fff405c 4022dc0f  
3fff3af0:  3fff49b4 00000001 3fff3b70 00000001  
3fff3b00:  3fff2678 3fff2640 3fff2678 40219662  
3fff3b10:  00000000 3fff2640 3fff3b60 40100690  
3fff3b20:  3fff2ac4 000004a3 00000000 4021eb48  
3fff3b30:  00000000 3fff2640 3fff26a4 fb0000e0  
3fff3b40:  00000000 3fff2640 3fff2678 3fff2640  
3fff3b50:  00000000 3fff2640 3fff2678 4021ab2a  
3fff3b60:  3fff6634 3ffec254 40219354 40219784  
3fff3b70:  00000000 00000000 402071dc 40207632  
3fff3b80:  3fff2678 3fff2aec 3fff25e0 4020aafb  
3fff3b90:  feefeffe feefeffe feefeffe feefeffe  
3fff3ba0:  feefeffe feefeffe feefeffe feefeffe  
3fff3bb0:  feefeffe feefeffe feefeffe 3fff2bac  
3fff3bc0:  3fffdad0 00000000 3fff2ba4 4021ed74  
3fff3bd0:  feefeffe feefeffe 3fff2bc0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

ESP8266 IR Controller
Config pin GPIO16 set to: 0
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
mounted file system
reading config file
opened config file
{"hostname":"ircontroller","passcode":"xxxx","port_str":"80","user_id":"amzn1.account.AFWWR5AFP2X2BHxxxxxxxxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.178.98
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
192.168.4.1
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
```

mdhiggins, 2018-01-06:
The change to a static IP address was done because the device needs port forwarding and would routinely have IP address changes (local IP) that would screw up the port forwarding. While using the MAC address to force a static IP in your individual router settings was possible, this was a bandaid solution so I moved to WiFiManagers static IP address configuration setup. It does not support dual mode so there was no way to leave dynamic options. That being said, static is the better option in this case.

Regarding your stack trace errors, this seems to happen when trying to write to the flash memory
I can't speak specifically to the WeMOS devices but I have had a couple NodeMCUs (probably about 3-4 out of the 20 or so I've used) that would frequently crash when trying to write to the flash memory. I sometimes found that disconnecting the NodeMCU from the breadboard during that saving operation helped and then returning it afterward got through the saving process but it was not consistent across the board. On those annoying devices I would just rerun wifiManager a few times until it saved without crashing then everything was good to go from there

Refreshing the blueprint won't erase the old settings
Not sure why your configpin to ground isn't working though, that should reset the device
During boot in the serial output it will announce the state of the configpin which unfortunately is omitted from the logs you've shared but that would be helpful

billcramer07, 2018-01-06:
Here is everything I did to get this running on Adafruit Huzzah ESP8266.
1. Change the reset pin (line 35) from 10 to 12. I no it's led 3 but it works. I only use 1 led.
2. Change line 46 to the static IP you want the esp to have. (ie 192.168.1.80).
3. Change line 47 to your gateway address. (ie 192.168.1.254).
Explanation;
GPIO 10 is used on the eeprom I think. I could get all the params to write to memory with wifi mgr EXCEPT the last 3 ip addresses. No matter how many times I would run wifi mgr, it would always try to connect with the values in  lines 46 and 47 no matter what I entered. Nor would it save the new addresses.
Hope this helps.
Bill C.

nicomania, 2018-01-07:
Hello Bill, hello mdhiggins,

the behavior of the configpin is really weird. It behaves differently on different pins:

```
Config pin GPIO10 set to: 1
```

```
Config pin GPIO16 set to: 0
```

Whatever.... At the moment I typed in my configuration inside the arduino code and compiled it. But it still refuses to connect as a client and opens an AP.

```
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.178.98
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
192.168.4.1
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
```

No matter how often I upload the code or how often I send a configuration via http, it does not connect. I can see my SSID in the webinterface, I click on it and I type in my password. That's how it worked at the start. It seems to me, it does not save my SSID and password. Where can I save them inside the code? I also do not see, that this is part of the config file:

```
opened config file
{"hostname":"ircontroller","passcode":"xxxx","port_str":"80","user_id":"amzn1.account.AFWWR5AFP2X2BHDxxxxxxxxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}
parsed json
```

By the way, right from the start I got a compiler error on line 381 saying, that the function is deleted:

```
Arduino: 1.8.5 (Windows 7), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 921600"
C:\xxxxx\Arduino\AlexaIRController\AlexaIRController.ino: In function 'bool setupWifi(bool)':
AlexaIRController:381: error: use of deleted function 'ESP8266WebServer& ESP8266WebServer::operator=(const ESP8266WebServer&)'
c:\xxxxx\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\unique_ptr.h:274:19: error: declared here
       unique_ptr& operator=(const unique_ptr&) = delete;
exit status 1
use of deleted function 'ESP8266WebServer& ESP8266WebServer::operator=(const ESP8266WebServer&)'
```

I then commented out this function call. As I use port 80, this should not make a difference for me and certainly not for my client connection problem.

```
379	  if (port != 80) {
380	    Serial.println("Default port changed");
381	//    server = ESP8266WebServer(port);
382	  }
```

mdhiggins, 2018-01-07:
I would restore back to the stock code since you've got multiple changes and its tough to debug with all those potential variables for problems

The issue with line 381 was due to a recent change with ESP8266 Arduino Core 2.4.0, you can roll back to 2.3.0 for now to fix this, I'll have a fix for 2.4.0 soon

The behavior of the config pin is expected, each GPIO pin has a default state and they are not all the same. GPIO10 for example defaults to 1 when not connected to anything, so depending on what pin you change to your config pin you need to make sure the default state isn't ground or it will continuously boot into configuration mode

nicomania, 2018-01-07:
My code is stock, except line 35 (configpin) and line 381 (mentioned function).

As I described, it tries to connect to a client and then it opens the AP because of failing. This has nothing to do with the configpin and this behavior does not change when switching back to the default pin.

mdhiggins, 2018-01-07:
Try rolling back to 2.3.0 and see if anything changes
I haven't been able to test anything on 2.4.0 yet I'm gonna try and get to it tonight when I get home

billcramer07, 2018-01-07:
Good morning
When I first tried this sketch and could not get anything to work with wifi mgr, I filled in the info between  the quotes on lines 40 thru 43 and also changed the static IP and gateway in lines 46 and 47 to my wifi params. After uploading the sketch it connected to my wifi and just worked. Maybe you can try that.

Bill C.

nicomania, 2018-01-07:
Yes, that's what I tried and also described above. But I doubt these settings will let your device connect to your wifi. It does not know which wifi to connect to, unless you fill in the configuration in your browser. Where did you type in the name (SSID) and password of your wifi?

mdhiggins, 2018-01-07:
Did you roll back to 2.3.0?

mdhiggins, 2018-01-07:
Also what type of WiFi security is your router using?

nicomania, 2018-01-07:
No, I am waiting for a new release of your code. My router uses wpa2 and as I wrote before, it already worked until I changed the amazon id.

billcramer07, 2018-01-07:
Nico, I think Mr Higgins is right. I'm running 2.3.0. Get no compile errors. I think my problems are coming from the fact that the Adafruit esp board I'm using is just not fully compatible. I think it has an issue writing to the eeprom. I do no for a fact that if I leave the wifi reset pin as default, it gets continuous wdt resets. As to how I finally got it on my wifi, it does seem to save the ssid and pwd in wifi mgr. The rest of the info I put in like I wrote earlier.

billcramer07, 2018-01-07:
BTW, I will have a Node MCU v.1 delivered this afternoon from Amazon. I'll let you know how it goes.

Bill C

billcramer07, 2018-01-07:
Ok, got the NodeMCU. Loaded a fresh copy of the sketch only changing the time zone offset. Booted into wifi mgr, input info and away it went. It did do one of those strange data dumps when I hit save but it booted and connected to my wifi and Alexa with no issues. Some ESP boards are just not compatible I guess.

mdhiggins, 2018-01-08:
I was randomly getting Exception 28 as well and it seems to be related to mDNS and an error that is thrown when the router wasn't responding appropriately
I did some updates that are now live that make things compatible with 2.4.0 so you should be good to update everything

That being said, while my Exception 28 has resolved, I think it was honestly restarting my router that fixed it cause the code changes I made shouldn't have impacted things

I ordered a WeMOS D1 to test on so hopefully I can pinpoint things but exception 28 seems to be mDNS related

mdhiggins, 2018-01-08:
mDNS changes
b2d935275259a945502850fcebd87cb8a52cf357

mdhiggins, 2018-01-10:
Looks like a lot of this is related to lwIP version 2.0
This became the default setting with ESP8266 Arduino Core 2.4.0

For now rolling back to lwIP 1.4 seems to fix everything
WiFiManager will likely need an update to address  this

Tools > lwIP > lwIP 1.4 (prebuilt)

closed 2018-01-10 by mdhiggins

nicomania, 2018-01-13:
Thank you for the update. I can confirm, Exception 28 is gone now with lwIP 1.4 and my wifi connects. But I need to configure it after every reset, because it does not seem to safe the settings.
So I tried to configure the rest, to get the whole to work with Alexa. But every time I try to send a command via http, I get: 
```
Web: "Error - last authentication failed because your timestamps are out of sync, see serial output for debugging details"
Serial: "Failed security check, signature is too old"
```
My timezone is set correctly and the timestamps in the web ui are correct. 
I am giving up on this. This does not seem to be a mature project. Thanks anyway for your help.
Best regards

mdhiggins, 2018-01-13:
Can you post the full serial output for that failure? It should post the actual timestamps

Most of the problems you were experiencing were introduced by problems with the ESP8266 Arduino update itself, if you look on their issues page they are trying to sort a fix as well, so its really not an issue with the project being mature but with a dependency change that's caused issues with multiple projects

Either way that info will help me improve the project for those in the future

nicomania, 2018-01-13:
Here you go:
```
ESP8266 IR Controller
Config pin GPIO2 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"ircontroller","passcode":"xxxx","port_str":"80","user_id":"amzn1.account.AFWWR5AFP2X2BHDxxxxxxxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.178.98
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
192.168.4.1
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: Scan done
*WM: DUP AP: Telekom_FON
*WM: GeneralagenturERGO
*WM: -76
*WM: WLAN-804741
*WM: -82
*WM: WLAN-Prusak
*WM: -82
*WM: Telekom_FON
*WM: -83
*WM: FRITZ!Box Fon WLAN 7340
*WM: -83
*WM: DIRECT-20-HP OfficeJet Pro 8720
*WM: -85
*WM: WLAN-470811
*WM: -86
*WM: ICON-1968b0
*WM: -89
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: ircontroller
*WM: Parameter
*WM: passcode
*WM: xxxx
*WM: Parameter
*WM: port_str
*WM: 80
*WM: Parameter
*WM: user_id
*WM: amzn1.account.AFWWR5AFP2X2BHxxxxxxxxxxxx
*WM: static ip
*WM: 192.168.178.98
*WM: static gateway
*WM: 192.168.178.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.178.98
*WM: Connection result: 
*WM: 3
Should save config
WiFi connected! User chose hostname 'ircontroller' passcode 'xxxx' and port '80'
 config...
{"hostname":"ircontroller","passcode":"xxxx","port_str":"80","user_id":"amzn1.account.AFWWR5AFP2X2BHxxxxxxxxxxxxxxxxxx","ip":"192.168.178.98","gw":"192.168.178.1","sn":"255.255.255.0"}
Writing config file
Config written successfully
WiFi configuration complete
Local IP: 192.168.178.98
URL to send commands: http://ircontroller.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to ircontroller.local:80
HTTP Server started on port 80
Turning off the LED to save power.
External IP: 178.7.xxx.xxx    //censored
External IP address request took 3142 ms
No errors detected with security configuration or access to external validation servers during startup
Ready to send and receive IR signals
Connection received - MSG
Failed security check, signature is too old
Server: 0
Local: 1515853470
MID: 
```
And web ui after sending the command:
`Unauthorized, HMAC security authentication`

mdhiggins, 2018-01-13:
So it looks like there's no timestamp sent with the request, was that a request coming from Alexa?

Once you set the user_id this enables all security features and outside requests not coming from Alexa are intentionally blocked, so regular HTTP requests won't work and don't contain all the needed information

"Device secured with SHA256 authentication. Only commands sent and verified with Amazon Alexa and the IR Controller Skill will be processed" should be displayed on the status page

mdhiggins, 2018-01-13:
And as far as the settings not being saved, are you by chance flashing your wemos without a filesystem enabled? That will certainly make saving settings difficult
The file system the ESP8266 uses is called SPIFFS

In Arduino IDE, Tools > Flash Size > 1M (512K SPIFFS)
Using 1M (no SPIFFS) will mean no space is allocated for the filesystem and you won't be able to save settings