mdhiggins/ESP8266-HTTP-IR-Blaster #118: Panasonic TV not responding

MarkRiddoch, 2018-01-07:
I'm trying to get the IR Blaster to work with my Panasonic TV and DVR, but failing to get the TV to respond, has anybody managed to make this work?

The codes I learn from the remote for On and Off of the two devices are:
code=40040100BCBD:PANASONIC:48&address=0x4004   for the TV
and
code=40040D00BCB1:PANASONIC:48&address=0x4004   for the DVR

Alexa is talking to the Blaster fine and the codes the blaster web pages shows as being sent are the same as I expect. I have tried using pulse and pdelay (upto 20 or more times) but no response from the TV (the IR receive light never flickers).

I replaced the IR LED with a visible light LED (and resistor to bring the current down) and that flickered as expected - so I am pretty sure the electronics are fine. I've tried a couple of different IR LED's with r different resistors to match the current they could take. Also tried with less current, and a fresh LED, in case I had overdone it. I tried ranges from 2 inches to 2 to 3 feet. Nothing I try makes the TV or video respond.

Any suggests would be gratefully received
Thanks
Mark

mdhiggins, 2018-01-07:
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/75

See if anything there is helpful regarding your codes

MarkRiddoch, 2018-01-08:
Thanks for getting back to me so quickly. I tried converting to use decimal, as indicated in the issue #75, but no change. I am beginning to suspect my IR LED's, so have order some of the Adafruit ones so that I know I have the same setup. Will try again when I get those.

mdhiggins, 2018-01-08:
Did you also try using the shorter codes that were listed there?
That person had some success removing the 40040 prefix from the codes (which seemed to be the address) and using the shorter version plus the address setting

MarkRiddoch, 2018-01-08:
Thank you, having read the issue I managed to somehow miss the bit about removing the 40040 prefix. Having done that things are working well, except for occasional failure to switch on. I think this is related to the number of times it needs to pulse the on command. I will play with this and hopefully make it more reliable.

Great project and thanks for the support.

mdhiggins, 2018-01-12:
Glad its working, gonna close for now

closed 2018-01-12 by mdhiggins