mdhiggins/ESP8266-HTTP-IR-Blaster #119: IR Controller Setup, channel controller

CSchoneweg, 2018-01-08:
Hi,

great project, congratulations! It´s a real alternative to Logitech´s Harmony Hub where the 
skill unfortunately supports only one hub.

However, I also have some problems with this:

1. the cannel controller

Base URL ('for the URL to the /json path of your IR Controller, include your passcode if you use one')
In this section I put in the following (in the IR Controller Setup routine): 
http://31.18.xxx.xxx:80/json?&pass=pass, 

for Channel Up this:
http://31.18.xxx.xxx:80/msg?code=807FC03F:NEC:32&repeat=*chan*&pass=pass 

and for the buttons, like for example the first Button,this: 
data:'807F6A95',type:'NEC',length:32 alternating '807F6A95','NEC',32.

But nothing happens. The answer from Alexa is "Device does not support this" (translated from 
the german message "TV Küche unterstützt das nicht"). Directly sent as http-code it works!
What is wrong?

2. the volume controller

Telling Alexa to step up the volume it works. Telling Alexa to step up the volume 3 (or other) steps
it doesn't. The serial monitor shows the command to Alexa was correct:

Turning off the LED to save power.
851:rc5:12
851:rc5:12
851:rc5:12
Transmission complete

But the volume steps up (or down) only one time and the command is transmittet only once as http 
monitor shows. Why?

This is the code: http://31.18.xxx.xxx:80/msg?code=850:RC5:12&rdelay=1000&repeat=*vol*&pass=pass

Hope you can solve my problems!

mdhiggins, 2018-01-08:
https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html

Looks like DE is not listed under the supported languages for the channel controller so its possible that Amazon is blocking this on their end, though without a german device to test on I can't say for sure but that's my best guess, though they appear to not listed DE on any of the new smarthome skills

Stupid debugging question but you did remember to check the 'enabled' checkbox for channel controller right? (on pythonanywhere)

As far as the volume goes, if you're seeing the command pop up 3 times then everything it working as intended; if your TV isn't registering the subsequent commands then try adjusting your rdelay setting it may just be a timing issue

CSchoneweg, 2018-01-08:
thank you so far, and yes I enabled the checkbox for channel controller. The skill and its description is in german and the german language is as described in the skill supportet. But of course I do not know if there are some restrictions/limitations for german. Described is nothing like that.

CSchoneweg, 2018-01-08:
I set the rdelay for volume control to 2500ms but the command is transmitted only once to the http server.
In the serial monitor it seems all correct, vol down as well as vol up.

mdhiggins, 2018-01-08:
If you use a camera to look at the IR led do you see it blink 3 times?

CSchoneweg, 2018-01-08:
It blinks only one time

CSchoneweg, 2018-01-08:
now I changed the volume commands from /msg? to /json? .  The answer from Alexa is something like this: "The device seem to function not properly" (tranlated from german: "Das Gerät scheint nicht richtig zu funktionieren"). Rechanging the commands to /msg? again the answer from Alexa is "OK". But the old problems are back again: only one command sent via http, only one blink but x-times in the serial monitor. Mute does not work neither with /json? nor with /msg?.

CSchoneweg, 2018-01-08:
By the way, I think there is a little mistake in your new sketch: "const bool getTime = false;"
If is not set to "true", you´ll never get the right time!

mdhiggins, 2018-01-08:
Its not set to false in the blueprint
`const bool getTime = true; `

And there are overrides where the getTime setting will be ignored for authentication purposes

mdhiggins, 2018-01-08:
Can you post whatever settings you're using that are giving you the 'not functioning properly' error?
That's a good sign actually it means the commands are making it through to the device, just that the formatting of your URL is probably wrong

CSchoneweg, 2018-01-08:
for example the code for volume up:
http://31.18.xxx.xxx:80/json?pass=pass&plain=[{'type':'RC5','data':'850','length':12,'rdelay':500}]

sorry for the getTime

CSchoneweg, 2018-01-08:
if I send the code 
http://31.18.xxx.xxx:80msg?pass=pass&code=850:RC5:12&repeat=4&rdelay=800
directly to my http server the result is th same: one blink, code only once transmittet

CSchoneweg, 2018-01-08:
if I send this
http://31.18.xxx.xxx:80/json?pass=pass&plain=[{'type':'NEC','data':'850','length':12,'repeat':4,'rdelay':800}]
result is "Success! Code sent", but only once!

billcramer07, 2018-01-08:
The single quotes are missing around the 800. Is that an issue?

mdhiggins, 2018-01-08:
http://blaster.local/msg?code=850:RC5:12&repeat=4&rdelay=800 works fine for me on my device
I get 4 blinks 800ms apart on camera

Maybe its a hardware issue? Feel free to post a pic of your build

mdhiggins, 2018-01-08:
http://blaser.local/json?plain=[{'type':'NEC','data':'850','length':12,'repeat':4,'rdelay':800}]
Also works with 4 blinks 800ms apart with my test hardware

mdhiggins, 2018-01-08:
To use the JSON variant of the command for volume this is the format you should use:

http://31.18.xxx.xxx:80/json?pass=pass&plain=[{'type':'RC5','data':'850','length':12,'rdelay':800,'repeat':*vol*}]

Also not sure if this is for testing or what but you're alternating between RC5 and NEC commands on your examples, might be intended

CSchoneweg, 2018-01-08:
The reason for changing the commands is that the RC5 is for TV and NEC is for a XORO set top box.
I only wantet to know which command is right /json? or /msg?. Therefore I testet various command with my browser to see what it shows. For the turn on command I use this:
http://31.18.xxx.xxx:80/json?pass=pass&plain=[{'type':'NEC','data':'807FFA05','length':32,'rdelay':1500},{'type':'RC5','data':'4C','length':12}]&device=TV Kueche&state=1
It works fine and results in in two blinks and two seen commands on the http server. And in real life the TV and the XORO are both turning to on and off, via amazon or via browser. So I don´t think it´s a hardware problem, using a wemos d1 r2 mini with exact the other parts of your shopping list. 

When I test commands with the browser I think I can´t use something like *vol* because this is filled by amazon?

[image]

mdhiggins, 2018-01-09:
Correct the vol value is filled in by Amazon, you don't need the *vol* thing if you're using /msg, you do need it for /json 
Are you using a transistor in your build?

You can use JSON or message, whichever they are interchangeable for single commands, but json is needed if you want to use multiple commands

CSchoneweg, 2018-01-09:
To see what´s going on I replaced the IR led with a normal green led on my breadboard. And oh wonder, the wemos blinks only once and on the http server is seen only one entry, but green led blinks x+1 times. I think your first guess that its a timing problem is correct. So this problem is nearly solved. So I close this issue until amazon supports german for the channel controller. Many thanks so far.

closed 2018-01-09 by CSchoneweg