mdhiggins/ESP8266-HTTP-IR-Blaster #120: Amazon ID

Lazerjules, 2018-01-08:
Hello

Great project, thanks for sharing. 
I almost have this working. Alexa keeps returning ‘not responding’ and the serial buffer has a notification that user_id is in the wrong format and should be amzn1.account.***

Would you be able to clarify what the amazon ID is please, I don’t think I have one. It appears in the WiFi manager as something to be setup, but no mention in the tutorial or notes what this is or I have missed it. 

Many thanks 

Can you tell me what the

benlye, 2018-01-08:
Sign into https://tehpsyc.pythonanywhere.com/ and scroll right to the bottom of the page.  The ID is literally the last line on the page.

Lazerjules, 2018-01-08:
Oh yeah! 

So sorry.

closed 2018-01-08 by Lazerjules