mdhiggins/ESP8266-HTTP-IR-Blaster #121: Is IRController code runnable as-is, or does it need changes for specific board?

jskubick, 2018-01-09:
I'm attempting to get IRController working with a HiLetGo ESP-12e ( https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 ). This is my first experience with ESP8266, but I have a fair amount of past experience with Arduino (Uno/Nano/Leonardo/Mega 2560 ADK). 

The ubiquitous Arduino "Blink" app compiles, builds, downloads, and seems to run just fine (the GPIO high/low appears to be inverted, but from what I've read, that's normal for ESP8266).

IRController is another matter. It seems to compile and upload without errors, but all I get is a blinking red LED (it's near the "RST" button) afterwards. The blue LED on the module briefly comes on for a fraction of a second after pressing RST, then goes off (I can't say for sure it never flashes again, but if it does, I haven't seen it). 

If I open SerialMonitor (115200 baud) and hit the RST button, I see the following:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
⸮T⸮bUUUUE⸮⸮Z⸮⸮բ⸮U⸮⸮⸮⸮⸮⸮⸮⸮⸮JPkZ⸮⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮-k⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮VK⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUT⸮U⸮ZEj⸮-EZ뢵⸮⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮E⸮j⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮Ej⸮-UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮V⸮⸮j*⸮⸮⸮U⸮ZQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮uT⸮E⸮UJ⸮⸮⸮TT⸮B⸮⸮Z⸮TUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮⸮U⸮ZE⸮U⸮UZ⸮VE⸮⸮⸮PVմZ+⸮Z⸮UUE⸮⸮Z⸮⸮ըVU⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZEj⸮-⸮⸮⸮U⸮⸮TTU⸮Z⸮J⸮⸮⸮⸮⸮⸮+Z⸮TQ⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮⸮jZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮VUP⸮⸮TZT⸮(E⸮⸮⸮⸮Q*U⸮ZV⸮*UE⸮QQQQ⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ⸮U⸮⸮⸮⸮⸮⸮⸮⸮⸮JPkZ⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮K⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUTQT⸮⸮⸮QT⸮⸮⸮TKVԵ⸮⸮ZQ⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮EVj⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮⸮Z⸮UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮ZQT⸮⸮⸮⸮VUR⸮V⸮⸮⸮Z⸮Z⸮Z⸮QQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮U⸮⸮⸮T⸮R⸮⸮UPԵZ⸮TUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮QT⸮⸮⸮⸮VUVV⸮⸮⸮R⸮Z⸮UVk⸮B⸮T⸮⸮⸮UBQ⸮ZkTUZE⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZ⸮Z⸮⸮⸮⸮U⸮⸮TTU⸮Z⸮J⸮⸮⸮⸮⸮⸮+Z⸮TQ⸮⸮⸮UTi+⸮⸮⸮⸮Ze+⸮T⸮P⸮⸮jjZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮VUP⸮⸮TZT⸮(⸮⸮⸮j*Q*UE⸮j*Q*U⸮(QQQ⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բVU⸮⸮⸮⸮⸮⸮⸮⸮⸮JPkZ⸮⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮-k⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮VK⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUT⸮U⸮Z⸮Z⸮EZ뢵⸮⸮ZQ⸮U⸮ZE⸮⸮⸮⸮⸮VE-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮EVj⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮⸮Z⸮UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮⸮Z⸮⸮⸮UT⸮⸮⸮⸮⸮⸮⸮⸮VUR⸮⸮⸮⸮⸮Z⸮ZEj⸮-QQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮U⸮⸮⸮TYR⸮⸮UP⸮V⸮V⸮⸮PԵU⸮Z+⸮⸮V⸮U⸮Z⸮Z⸮E⸮U⸮UZ⸮VE⸮⸮⸮PVմZ+Ej⸮-UUE⸮⸮Z⸮⸮ը⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTe*ժZ⸮Z⸮⸮⸮⸮U⸮⸮TTU⸮ZQ⸮⸮P⸮Z⸮Z⸮+Z⸮T⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮jjZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮⸮UP⸮⸮TZT⸮(⸮⸮⸮j*Q*U⸮ZV⸮*U⸮(QQQ⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ⸮U⸮⸮⸮⸮⸮⸮⸮⸮⸮JP⸮Z⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UU⸮Z⸮E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮K⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮Ej⸮-խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUT⸮U⸮Z⸮Z⸮EZ뢵⸮⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮EVjZUVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮⸮Z⸮UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮ZQT⸮⸮⸮Ե⸮⸮Zյ⸮⸮⸮T⸮⸮⸮U⸮ZQQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮U⸮⸮⸮TYR⸮⸮UP⸮V⸮⸮⸮⸮PԵU⸮Z+⸮⸮V⸮U⸮ZEj⸮-E⸮U⸮UZ⸮VE⸮⸮⸮P⸮մZ+⸮Z⸮UUE⸮⸮Z⸮⸮ը⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZEj⸮-⸮⸮⸮U⸮⸮TTU⸮ZQ⸮⸮P⸮Z⸮Z⸮+Z⸮T⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮⸮jZ⸮⸮⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮VUP⸮⸮TZT⸮(⸮⸮⸮j*Q*U⸮ZV⸮⸮*UE⸮QQQQ⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ⸮U⸮⸮⸮⸮⸮⸮⸮⸮⸮JPkZ⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮K⸮KZ⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮V⸮⸮U⸮⸮⸮⸮⸮⸮(⸮RUTQT⸮⸮⸮QT⸮⸮⸮TKVԵ⸮⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮E⸮j⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮Ej⸮-UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮ZQT⸮⸮⸮Ե⸮⸮Zյ⸮⸮⸮T⸮⸮⸮U⸮ZQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮uT⸮E⸮UJ⸮⸮⸮TT⸮B⸮⸮ZkTUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮⸮U⸮ZE⸮U⸮UZ⸮VE⸮⸮⸮PVմZ+⸮Z⸮UUE⸮⸮Z⸮⸮ը⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZEj⸮-⸮⸮⸮U⸮⸮TTU⸮Z⸮J⸮⸮⸮⸮⸮⸮+Z⸮TQ⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮jjZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮⸮ZU+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮⸮UP⸮⸮TZT⸮(EJ⸮⸮⸮Q*U⸮ZV⸮*UE⸮QQQQ⸮/⸮⸮T⸮⸮UUUE⸮⸮Z⸮⸮բVU⸮⸮⸮⸮⸮⸮⸮⸮⸮JPkZ⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UU⸮Z⸮E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮VK⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUTQT⸮⸮⸮QT⸮⸮⸮TK⸮⸮VUR⸮⸮U⸮ZE⸮⸮⸮⸮⸮VE-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮E⸮jZUVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮Ej⸮-UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮⸮Z⸮⸮⸮UT⸮⸮⸮⸮⸮⸮⸮Ե⸮⸮Z⸮V⸮⸮j*⸮⸮QT⸮⸮⸮⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮UE⸮UJ⸮⸮TT⸮BQ⸮Z⸮TUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮⸮U⸮ZE⸮U⸮UZ⸮Ve⸮⸮⸮P⸮մZ+⸮Z⸮UUE⸮⸮Z⸮⸮ըVU⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTe*ժZEj⸮-⸮⸮⸮U⸮⸮TTU⸮ZQ⸮⸮P⸮Z⸮Z⸮+Z⸮TQ⸮
```

Current settings in Arduino IDE (v1.8.5):
* Board: NodeMCU 1.0 (ESP12e module)
* Flash Size: 4M (3M SPIFFS)
* Debug Port: Disabled
* Debug Level: None
* IwIP Variant: v2 Prebuilt (MSS=536)
* CPU Frequency: 80MHz
* Upload Speed: 921600
* Port: COM7
* Get Board Info returns "BN: Unknown board; VID: 10C4; PID: EA60; SN: Upload any sketch to obtain it" (I have a hunch this is what it's supposed to return for ESP12e boards since the IDE was originally written for AVR-based Arduinos, but I'm not 100% sure)

The settings above are either Arduino IDE defaults or were taken from the board's listing at Amazon. 

If it matters... the actual module says, "Model ESP8266MOD" and "Vendor: AI-THINKER". The back of the board says "NodeMCU Amica" and mentions CP2102 (apparently, other variants of "NodeMCU either don't have a dedicated USB-serial bridge of their own, or use different chips). I have no idea whatsoever whether the CP2102 or ESP8266 module are "genuine", or whether they flawlessly implement the "Amica" reference design. 

Things I've tried so far:

* Disconnecting everything but the USB cable from the ESP12e board

* Attaching an external regulated 5v power supply to Vin and Gnd

* Adding capacitors across Vin and Gnd (I tried .1uf ceramic, 2.2uF tantalum, and a 47uF electrolytic).

* Searching for a nearby wifi access point (I got the impression from the project's instruction video that it emulates an AP if you press RST while grounding GP10(SD3). 

The board appears to be periodically outputting something to serial monitor... but it definitely isn't anything that resembles English text at 115200, 1mbps, 300, 1200, 2400, 4800, 9600, 19200, 38400, 57600, or 74880 baud.

I'm using Arduino IDE 1.8.5. I vaguely recall seeing a few comments somewhere about "recent" versions of the Arduino IDE having problems with ESP8266... but I'm not sure just how "recent" is considered "recent"... or whether it's even still a problem.

Does any of this look like a problem you've seen before?

closed 2018-01-09 by jskubick

mdhiggins, 2018-01-09:
You can  try adjusting the blueprints pinout configuration to see if that helps

Try changing the configpin to something besides the default setting (can use one of the other reserved 4 output pins) and see if that helps at all

If not you can try moving around the output and receiver pins and see if that makes any difference

I’ve already read that people can get that kind of performance if your USB power source isn’t outputting enough current
If you have a different USB adapter that you can try that’s probably a good thing to do as well

From: Jeff Skubick
Sent: Tuesday, January 9, 2018 3:06 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] Is IRController code runnableas-is, or does it need changes for specific board? (#121)

I'm attempting to get IRController working with a HiLetGo ESP-12e ( https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 ). This is my first experience with ESP8266, but I have a fair amount of past experience with Arduino (Uno/Nano/Leonardo/Mega 2560 ADK).
The ubiquitous Arduino "Blink" app compiles, builds, downloads, and seems to run just fine (the GPIO high/low appears to be inverted, but from what I've read, that's normal for ESP8266).
IRController is another matter. It seems to compile and upload without errors, but all I get is a blinking red LED (it's near the "RST" button) afterwards. The blue LED on the module briefly comes on for a fraction of a second after pressing RST, then goes off (I can't say for sure it never flashes again, but if it does, I haven't seen it).
If I open SerialMonitor (115200 baud) and hit the RST button, I see the following:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
⸮T⸮bUUUUE⸮⸮Z⸮⸮բ�⸮U⸮⸮⸮⸮⸮⸮⸮⸮�⸮JPkZ⸮⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮-k⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮VK⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮�խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUT⸮U⸮Z�Ej⸮-EZ뢵⸮⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮E⸮j⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮�Ej⸮-UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮V⸮⸮j*⸮⸮⸮U⸮ZQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮uT⸮E⸮UJ⸮⸮⸮TT⸮B⸮⸮Z⸮TUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮⸮U⸮ZE⸮U⸮UZ⸮VE⸮⸮⸮PVմZ+�⸮Z⸮�UUE⸮⸮Z⸮⸮ըVU⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZ��Ej⸮-⸮⸮⸮U⸮⸮TTU⸮Z⸮J⸮⸮⸮⸮⸮�⸮+Z⸮TQ⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮⸮jZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮VUP⸮⸮TZT⸮(�E⸮⸮⸮⸮Q*U⸮ZV⸮�*UE⸮�QQQQ⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ�⸮U⸮⸮⸮⸮⸮⸮⸮⸮�⸮JPkZ⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮K⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮�խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUTQT⸮⸮⸮QT⸮⸮⸮TKVԵ⸮⸮ZQ⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮EVj⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮�⸮Z⸮�UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮ZQT⸮⸮⸮⸮VUR⸮V⸮⸮⸮Z⸮Z�⸮Z⸮�QQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮U⸮⸮⸮T⸮R⸮�⸮UPԵZ⸮TUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮QT⸮⸮⸮⸮VUVV⸮⸮⸮R⸮Z�⸮UVk⸮B⸮T⸮⸮⸮UBQ⸮ZkTUZE⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZ��⸮Z⸮�⸮⸮⸮U⸮⸮TTU⸮Z⸮J⸮⸮⸮⸮⸮�⸮+Z⸮TQ⸮⸮⸮UTi+⸮⸮⸮⸮Ze+⸮T⸮P⸮⸮jjZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮VUP⸮⸮TZT⸮(�⸮⸮⸮j*Q*UE⸮j*Q*U⸮(QQQ�⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ�VU⸮⸮⸮⸮⸮⸮⸮⸮�⸮JPkZ⸮⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮-k⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮VK⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮�խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUT⸮U⸮Z�⸮Z⸮�EZ뢵⸮⸮ZQ⸮U⸮ZE⸮⸮⸮⸮⸮VE-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮EVj⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮�⸮Z⸮�UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮⸮Z⸮⸮⸮UT⸮⸮⸮⸮⸮⸮⸮⸮VUR⸮⸮⸮⸮⸮Z⸮Z�Ej⸮-QQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮U⸮⸮⸮TYR⸮�⸮UP⸮V⸮V⸮⸮PԵU⸮Z+⸮⸮V⸮U⸮Z�⸮Z⸮�E⸮U⸮UZ⸮VE⸮⸮⸮PVմZ+�Ej⸮-UUE⸮⸮Z⸮⸮ը⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTe*ժZ��⸮Z⸮�⸮⸮⸮U⸮⸮TTU⸮ZQ⸮⸮P⸮Z⸮Z⸮+Z⸮T⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮jjZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮⸮UP⸮⸮TZT⸮(�⸮⸮⸮j*Q*U⸮ZV⸮�*U⸮(QQQ�⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ�⸮U⸮⸮⸮⸮⸮⸮⸮⸮�⸮JP⸮Z⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UU⸮Z⸮�E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮K⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮Ej⸮-խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUT⸮U⸮Z�⸮Z⸮�EZ뢵⸮⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮EVjZUVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮�⸮Z⸮�UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮ZQT⸮⸮⸮Ե⸮⸮Zյ⸮⸮⸮T⸮⸮⸮U⸮ZQQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮U⸮⸮⸮TYR⸮�⸮UP⸮V⸮⸮⸮⸮PԵU⸮Z+⸮⸮V⸮U⸮Z�Ej⸮-E⸮U⸮UZ⸮VE⸮⸮⸮P⸮մZ+�⸮Z⸮�UUE⸮⸮Z⸮⸮ը⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZ��Ej⸮-⸮⸮⸮U⸮⸮TTU⸮ZQ⸮⸮P⸮Z⸮Z⸮+Z⸮T⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮⸮jZ⸮⸮⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮u⸮U+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮VUP⸮⸮TZT⸮(�⸮⸮⸮j*Q*U⸮ZV⸮⸮*UE⸮�QQQQ⸮/⸮ŢT⸮bUUUUE⸮⸮Z⸮⸮բ�⸮U⸮⸮⸮⸮⸮⸮⸮⸮�⸮JPkZ⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UUEj⸮-E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮K⸮KZ⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮V⸮⸮U⸮⸮⸮⸮⸮⸮(⸮RUTQT⸮⸮⸮QT⸮⸮⸮TKVԵ⸮⸮Z⸮U⸮ZE⸮⸮⸮⸮⸮Ve-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮E⸮j⸮UVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮�Ej⸮-UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮uj*⸮⸮⸮UZ⸮ZQT⸮⸮⸮Ե⸮⸮Zյ⸮⸮⸮T⸮⸮⸮U⸮ZQ⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮uT⸮E⸮UJ⸮⸮⸮TT⸮B⸮⸮ZkTUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮⸮U⸮ZE⸮U⸮UZ⸮VE⸮⸮⸮PVմZ+�⸮Z⸮�UUE⸮⸮Z⸮⸮ը⸮U⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTE*ժZ��Ej⸮-⸮⸮⸮U⸮⸮TTU⸮Z⸮J⸮⸮⸮⸮⸮�⸮+Z⸮TQ⸮⸮⸮UTi+⸮⸮⸮⸮ZE+⸮T⸮P⸮⸮jjZ⸮*⸮P*V⸮⸮ET⸮⸮⸮⸮T⸮⸮⸮*⸮⸮ZU+UZ⸮⸮UT⸮⸮⸮TQT⸮UQ⸮⸮jP⸮,⸮⸮⸮UP⸮⸮TZT⸮(�EJ⸮⸮⸮Q*U⸮ZV⸮�*UE⸮�QQQQ⸮/⸮⸮T⸮⸮U�UUE⸮⸮Z⸮⸮բ�VU⸮⸮⸮⸮⸮⸮⸮⸮�⸮JPkZ⸮Q⸮⸮UV⸮TU⸮⸮⸮W⸮⸮⸮⸮U⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮uZk⸮UU⸮Z⸮�E*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮⸮⸮R⸮⸮TU⸮ZE*UV⸮⸮ZTU⸮⸮⸮⸮⸮⸮⸮VK⸮KZQT⸮⸮⸮⸮T⸮V⸮T⸮U⸮⸮⸮⸮Z⸮�խ⸮W⸮⸮⸮j⸮⸮⸮⸮(⸮RUTQT⸮⸮⸮QT⸮⸮⸮TK⸮⸮VUR⸮⸮U⸮ZE⸮⸮⸮⸮⸮VE-m⸮⸮⸮U⸮⸮VZ⸮T⸮⸮⸮T⸮⸮UP⸮⸮⸮E⸮jZUVVJ⸮⸮⸮⸮⸮⸮+⸮j⸮�Ej⸮-UZ⸮V⸮T⸮⸮T⸮⸮⸮TV⸮*⸮⸮Z⸮⸮⸮UT⸮⸮⸮⸮⸮⸮⸮Ե⸮⸮Z⸮V⸮⸮j*⸮⸮QT⸮⸮⸮�⸮U⸮ZUR
ETU⸮TTR⸮⸮⸮T⸮⸮U⸮⸮⸮⸮UE⸮UJ�⸮⸮TT⸮BQ⸮Z⸮TUZE⸮U⸮UZ⸮⸮⸮⸮T⸮⸮⸮⸮U⸮ZE⸮U⸮UZ⸮Ve⸮⸮⸮P⸮մZ+�⸮Z⸮�UUE⸮⸮Z⸮⸮ըVU⸮UZ⸮⸮⸮⸮⸮U⸮ZETUTe*ժZ��Ej⸮-⸮⸮⸮U⸮⸮TTU⸮ZQ⸮⸮P⸮Z⸮Z⸮+Z⸮TQ⸮
Current settings in Arduino IDE (v1.8.5):
• Board: NodeMCU 1.0 (ESP12e module)
• Flash Size: 4M (3M SPIFFS)
• Debug Port: Disabled
• Debug Level: None
• IwIP Variant: v2 Prebuilt (MSS=536)
• CPU Frequency: 80MHz
• Upload Speed: 921600
• Port: COM7
• Get Board Info returns "BN: Unknown board; VID: 10C4; PID: EA60; SN: Upload any sketch to obtain it" (I have a hunch this is what it's supposed to return for ESP12e boards since the IDE was originally written for AVR-based Arduinos, but I'm not 100% sure)
[... cut ...]
The board appears to be periodically outputting something to serial monitor... but it definitely isn't anything that resembles English text at 115200, 1mbps, 300, 1200, 2400, 4800, 9600, 19200, 38400, 57600, or 74880 baud.
I'm using Arduino IDE 1.8.5. I vaguely recall seeing a few comments somewhere about "recent" versions of the Arduino IDE having problems with ESP8266... but I'm not sure just how "recent" is considered "recent"... or whether it's even still a problem.
Does any of this look like a problem you've seen before?

jskubick, 2018-01-23:
For what it's worth, it looks like my original ESP8266 board **was** defective after all... setting the serial baudrate to 74880 (while keeping serial monitor set to 115200) seemingly fixed the serial monitor problem, but nothing I did was able to make the wi-fi work (with this project, or with anything else). I finally threw in the towel and bought another ESP8266 module. The new board has the same 74880-vs-115200 baud issue as the first, but unlike the first board, the wifi worked perfectly on the first try.