mdhiggins/ESP8266-HTTP-IR-Blaster #122: Exception 28 after WiFiManager Configuration lwIP 2.0

mdhiggins, 2018-01-10:
Bug with lwIP 2.0 and WiFiManager that causes exception 28 and reboot when trying to save WiFiManager settings with staticIP configuration

Likely a limitation of WiFiManager and will require an update but listing this as a known issue

Current fix recommendation is to use lwIP version 1.4 (indicated on readme)

eukiseioh, 2018-02-09:
Ok, i used 2.0 and solved writting ip setting into ircontroller.ino file.
I will back compile and upload the file with 1.4 configuration. 👍

jebbett, 2018-03-15:
Adding a comment here to help others the ESP8266 V2.4.1 package has a memory leak in all versions of lwip 1.4 and 2.0 variants, so use the 2.4.0 package until a fix is put in place.

mdhiggins, 2018-03-29:
Yeah 2.4.1 isn't working with either variant so recommending 2.4.0 with lwIP 1.4 for the moment until they get thing sorted

closed 2020-03-20 by mdhiggins