mdhiggins/ESP8266-HTTP-IR-Blaster #123: Possible to not use Amazon Alexa and only Homebridge ?

jmaugerdumais, 2018-01-11:
Hello !
Nice plugin ! Is it possible to configure it to only use it through Homebridge - Homekit ? 

If so, how ? I only get "Unauthorized, HMAC security authentication failed" error ...

Thanks !

Julien

mdhiggins, 2018-01-11:
Not super familiar with home bridge but to disable to HMAC authentication you just need to leave the user_id setting blank in WiFiManager

neo7530, 2018-01-11:
I'm writing an WebApplication atm. It takes a moment, but when its ready, you can use a web browser or even a smartphone (native webapp) to control your devices. It will be written in PHP, so it runs on apache webserver (e.g. RaspberryPi). It is well documented, so you can tune it for your own needs. You can leave your device as it is (amazon-id/passcode) HMAC is calculated in PHP, so it will respond properly. When it is done, i will push it here(?). mdhiggins, can i upload it anywhrere, so you can put it in the wiki?

Marco.

jmaugerdumais, 2018-01-12:
It worked ! :) Thanks !

mdhiggins, 2018-01-12:
Great, gonna close this
@neo7530 I actually have a solution that I create a month ago on the pythonanywhere site that does what you're indicating, you sort of just push the commands through that but I hadn't polished it or made it live
If you wanna send me what you've got so far I can take a look

closed 2018-01-12 by mdhiggins