mdhiggins/ESP8266-HTTP-IR-Blaster #124: WeMOS pinouts

mdhiggins, 2018-01-12:
The default pinouts do not work on certain WeMOS devices, most commonly the D1 Mini
The configpin in particular does not work

If we can crowd source a list of working pinouts I'll add an alternative blueprint option for WeMOS devices

WeMOS users please share what pin configuration you're using

jonathanhickford, 2018-02-02:
I'm using a WeMOS D1 Mini (v2.3.0). I've not tried using multiple IR receivers, but the below work for me. 
```
const int pinr1 = D3;       // Receiving pin (Arduino GPIO0)
const int pins1 = D4;       // Transmitting preset 1 (GPIO2)
const int configpin = D2;   // Config pin (GPIO4)
```
In an ideal WeMOS world the config pin would perhaps be D3 (which is one used on the 1 button shield), but I had problems using that as the config pin.

vkhurana, 2018-02-04:
@jonathanhickford i tried this with my wemos, and it seems like the IR LED is always on? It only seems to go off when the wemos LED comes on. Can you share the circuit diagram you used? I'm using the one from readme, but adapted for Wemos D1.

jonathanhickford, 2018-02-07:
It was pretty much as the circuit diagram. I used the 5V output for the R2 -> IR LED -> collector.

lowtolerance, 2018-02-07:
D4 is what I used originally for transmission, and I couldn't figure out why it caused the LED to be so much dimmer than the same circuit hooked up to a NodeMCU on the same pin. Turns out that D4 is what the NodeMCU uses for it's onboard LED, and it already has a 10k resistor tied to that pin, so you end up with too much resistance on the line for full output. 

D3 and D8 also have a 10k resistor, leaving D0-D2 + D5-D8 as the most sensible choices for IO.

jonathanhickford, 2018-02-08:
Thanks. I'm now trying the below, however like @vkhurana I'm finding the LED remains on all the time.
```
const int pinr1 = D5;        // Receiving pin (GPIO14)
const int pins1 = D6;        // Transmitting preset 1 (GPIO12)
const int configpin = D2;    // Reset Pin (GPIO4)
```

vkhurana, 2018-02-09:
@jonathanhickford maybe i have the same issue - you're using a 2N222?

jonathanhickford, 2018-02-09:
Yep 2N222 now - and the resistor values and LED as per the readme. That resolved my problem.

mdhiggins, 2018-02-09:
Are you guys just eliminating the configuration for S2-S4 or just leaving them at default and not using them? Gonna try and add this to the wiki page for future users

lowtolerance, 2018-02-11:
I’ve just been leaving the unused declarations at their default values. Might not be a bad idea to be able to enable/disable the additional transmitters through a simple Boolean, but it’s hardly necessary.

jebbett, 2018-03-18:
const int pinr1 = 14;                                         // Receiving pin (D5)
const int pins1 = 4;                                          // Transmitting preset 1 (D2)
const int configpin = 3;                                   // Reset Pin (RX)

I've got a couple of D1 Minis using the above config, shame there isn't somewhere for people to upload their projects.. The mini has a nice prototype shield that's idea for this and keeps it tiny, I have one with IR leds built in and one using a Samsung Smart TV IR Blaster BN96-31644A, which consists of 5 IR LEDs pointing each in different directions, all in parallel.. giving 360x180 degree coverage..

closed 2020-04-24 by mdhiggins