mdhiggins/ESP8266-HTTP-IR-Blaster #125: Web-Blaster ;)

neo7530, 2018-01-13:
[remote.zip](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/1628276/remote.zip)
PHP-Web-Remote. Extract on a webserver... edit index.php for your own needs. A mapping-editor is inside also.

have fun, neo7530.

mdhiggins, 2018-01-13:
The mapgen application you've included in this zip contains a cryptocurrency miner (remote/mapgen/js/coinhive.js), download link removed

I don't know if any malice on your part, it looks like this was just part of the mapgen project but just want you to be aware

mdhiggins, 2018-01-13:
Trojan:HTML/Brocoiner!rfn
https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Trojan:HTML/Brocoiner!rfn

closed 2018-01-13 by mdhiggins

neo7530, 2018-01-13:
Just delete the coinminer, i have deactivated it in the main php. it was just a addon from the original mapmaker...

mdhiggins, 2018-01-13:
I would repackage and repost, chrome flagged the zip and wouldn't download and when I bypassed to take a look and then Windows caught it too and quarantined, I suspect most users will be uneasy and not give it a chance if that happens

neo7530, 2018-01-13:
[remote.zip](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/1628429/remote.zip)
Repacked without coinminer...