mdhiggins/ESP8266-HTTP-IR-Blaster #126: Connect WiFi after few minutes shutdown (not respond)

kleusbalut, 2018-01-13:
i've encountered this issue.

connect to WiFi, after few minutes to shutdown (not respond, no wdt logs)

modules different shutdown time.
my tested modules case, 2-4 min or 30 min to shutdown.

[Environment]
Module: ESP-WROOM-02
Arduino-IDE: 1.6.13
ESP8266-Arduino-Core: 2.4.0
OS: Windows10 x64 FallCreatorsUpdate

I'm tried below.

[Hardware Side]
- tests other module, gets same issue.
- Add capacitor (470uF)
- Change power source

[Software Side]
- Upgrade ESP8266-Arduino-Core to newer version (2.4.0)
- Config pin change to GPIO12 (in ESP-WROOM-02, GPIO10 is used on board chip communication)

any idea?

best regards.

mdhiggins, 2018-01-13:
Can you post  the serial output for these connection failures?

There is a reset feature built in if it gets stuck in AP mode for too long to help prevent devices that reboot from a power outage from getting stuck in AP mode perhaps that is what's triggering things

kleusbalut, 2018-01-14:
thank you for replying.

i've got this issue, after WiFi Configured and STA mode connection.
and no Serial output logs. (silent death, no reboot)

[Additional Information]
- WiFi Manager Settings seems good work. done connect to my WiFi Network.
- Arduino IDE setting to Debug Port: Serial, Debug Level: CORE+WIFI+HTTP_UPDATE+UPDATER+OTA

mdhiggins, 2018-01-14:
Hm with no serial output I wonder if one of the pinouts is conflicting, that is usually the issue
I would try switching around the pins and run the device disconnected from all other hardware except USB and see if you can get serial output to work

Usually when you're getting crashes and no serial output its because one of the GPIOs that the device is using is reserved for a system function

closed 2018-01-24 by mdhiggins