mdhiggins/ESP8266-HTTP-IR-Blaster #128: Static IP config does not work with lwIP 1.4

scop, 2018-01-14:
Commit 564c5f668d519875223ef8ce148fff20ca489652 breaks static IP config for me with lwIP 1.4, the configured values do not get applied. Commenting out the line added in this config fixes it for me.

mdhiggins, 2018-01-17:
That line was changed in a later update to WiFi.config(sip, sgw, sgw, ssn);

Did you have that update?

scop, 2018-01-17:
Yes I did, that's the one I have commented out. I don't think I tested the older one.

mdhiggins, 2018-01-17:
I can revert this for now as it was a bandaid fix for 2.0 which really shouldn't be supported now anyway due to the issues introduced. I haven't personally had issues with this but I think my router may be reverted to DHCP which still assigns the correct IP cause I have hard coded the IP's based on the MACs from before settings were static

mdhiggins, 2018-01-17:
395f2a581961e65d3771d181dbd15d5dff8c5aa2

closed 2018-01-17 by mdhiggins