mdhiggins/ESP8266-HTTP-IR-Blaster #138: Issue compiling with ESP8266 Core v2.3.0

benlye, 2018-01-18:
Since https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/commit/8fb3e70a67d658e13a368460dd763394ad8c947d, and its predecesor in #131, when compiling with v2.3.0 of the ESP8266 core the compiler gives an error:
```
C:\Users\blye\AppData\Local\Temp\arduino_build_749561\sketch\IRController.ino.cpp.o: In function `rokuCommand(String, String)':

C:\Users\blye\GitHub\ESP8266-HTTP-IR-Blaster\IRController/IRController.ino:232: undefined reference to `strtoull'

C:\Users\blye\GitHub\ESP8266-HTTP-IR-Blaster\IRController/IRController.ino:232: undefined reference to `strtoull'

collect2.exe: error: ld returned 1 exit status
```

The solution is to update the ESP8266 core to v2.4.0 (then ensure that lwIP is set to `lwIP 1.4 (Prebuilt)`).

Just figured I'd share, as there seems to have been some confusion (#132).  

Maybe the readme could be updated to say that the minimum core version required is 2.4.0?

zobotek, 2018-01-19:
Yes.  I had this problem and thought that strtoull was a typo, since strtoul (as it was before a pull) compiled.

When the author of Pull Request 131 noted that this was not a typo, I figured library update(s) were needed.  After using the latest version of all libs and Arduino, I found that lines with strtoull would compile.  

But then there were other issues:  HTTPClient DNS broke, and external IP address and time retrieval fail with lwIP 2.0, but work with lwIP 1.4 (Prebuilt).  The notes on github indicate that there are other potential issues with lwIP2.0.  Perhaps it would be useful if the notes on github list all a set of versions of all libraries (not just lwIP) that are known to work.

mdhiggins, 2018-01-19:
It’s a issue with lwIP that there in the midst of fixing per their GitHub. Otherwise the aim will always be to support the latest versions of the dependencies. I did note in the readme this one exception

zobotek, 2018-01-23:
FYI,  there’s a 3D printable case design for the @raptordemon medium PCB at https://github.com/zobotek/ESP8266-HTTP-IR-Blaster/blob/master/IRControllerCaseRaptorDMedV1_5.stl