mdhiggins/ESP8266-HTTP-IR-Blaster #140: ESP-01 Support?

quark72, 2018-01-22:
Hello!

What is neccessary to compile the Example for ESP-01 with 1MB?
Directly compile without changes bring the ESP-01 in a reboot loop. Doens't matter what is set for configpin.

How can be debuged where the Problem with ESP-01 is?

Thanks!

mdhiggins, 2018-01-24:
Not sure as I don't have one of those to test with and have only ever done development on the nodeMCU

I would try adjusting the pinout configuration as I suspect that's the main culprit but I can't say for sure

quark72, 2018-02-04:
Hello!

You are absolut right. The Proble was the configpin. I set it to 14 - something unusable, but i don't need it, and because there are not enough GPIO impossible too to use. Well it is possible to remove the Receive so to have a Reset-Pin... It is a Option. But Reset And IR-Receive and Send is not possible, in IMHO.
Theoreticaly RX and TX can be missused for Reset, but then the Serial Output sholud be disabled.

IR-Receiver i set and connected to GPIO 0, Send-LED to GPIO0.
Other change i make was, to use a N-Mosfet like 2N7000, because a normal NPN-Transistor works like pull-down resistor on GOIP2 and then ESP-01 doesn't boot. With 2N7000 the problem is gone. (I removed the Basis Resistor and connected the Gate directly to the ESP)
On GPIO0 i must connect a 10k Pull-Up Resistor - without the ESP-01 would not boot too (because the IR-Recevier act as Pull-Down).

I'm not sure if this is the best Solution to use a cheap ESP-01 but it works. 

Ofcourse there are disadnvatgaes:
- no Reset Pin like described
- Serial connection stop to work after connect to WiFi(?!). Maybe because at ESP-01 GPIO1 is in the same time TX. And in this Moment when it is set Serial connection stops working. It is enough to see, that the setup process is successfully. So for me it is ok so.

This is my more or less quick and dirty solution. May be somone is interesting in this descitption to use the very cheap ESP-01, which is more then enough for IR-Blaster. ;-)

Greetings!

mdhiggins, 2018-02-09:
Thanks for sharing
If you want to organize all the changes you made into a bullet list I'll put it on the wiki for others trying to get it to work with the cheaper ESP-01

closed 2018-02-16 by mdhiggins

nixnotwin, 2024-02-21:
I was able to install on Esp-01 after changing pinout configuration. After configuring it to to connect to my wifi router, it goes into boot loop. I had set reset pin to pin 14 because I wanted to use available 2 gpio's for reciever and sender (gpio 0 & 2).  Serial output says pin 14 was set to pin 1. But, pin 1 & 3 are required for serial communication. Is there anyway I can make it work on esp-01.