mdhiggins/ESP8266-HTTP-IR-Blaster #142: Erasing NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E

thommy6102, 2018-02-03:
Hello,

I`m using such a board:
NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E WIFI Wifi Development Board
https://www.amazon.de/gp/product/B06Y1ZPNMS/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

I followed your instruction, but I think I did a mistake inside the WiFi configuration.
In the Serial Monitor I`ve got this answer:

ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"","passcode":"","port_str":"80","user_id":"","ip":"10.0.1.10","gw":"10.0.1.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'
WiFi configuration complete
Local IP: 10.0.1.10
URL to send commands: http://ESP_ACA246.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to ESP_ACA246.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 5002 ms
Ready to send and receive IR signals

Every attempt to re-program the modul, give me the same result. I`ve got still the same Output as above
in the Serial Monitor.
Is there a possibility to erase (master reset) the modul, in order I can re-start from the beginnig? 

Many thanks in advance for any Support.

Greetings from Germany
Thomas

mdhiggins, 2018-02-05:
You just need to set the configpin to ground
`Config pin GPIO10 set to: 1` (needs to be 0, will boot into config mode)
GPIO10 is the default configuration pin

eukiseioh, 2018-02-09:
I have the same problem but i edited ircontroller.ino file and writed ip serttings:

char static_ip[16] = "10.0.1.10";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

Change “10.0.1.10” static ip to valid ip LAN like 192.168.xxx.xxx

Change “static gw” to your router gw ip

Save and upload again to board. For webroot access target to static ip LAN.

mdhiggins, 2018-02-09:
I would advise against hard coding as you're just band-aiding what is probably an actual issue as @eukiseioh recommends here

Make sure when you compile your project that SPIFFS is enabled and allocated (not 0) and that when you're selecting your board you are using the correct one

If you find its still giving you a hard time saving things to system memory you can try changing around your GPIO pins (probably the configuration pin) and see if maybe there's a conflict with your particular board that's preventing writing to the file system while one of those pins is allocated

thommy6102, 2018-02-10:
It seems that my problem has to do with the revision number of my NodeMCU.
I`ve got version V3 and that seems to be little different. After I flashed it for the very first time, I could see
the AP (192.168.4.1), connect to it with a Android Tab, and make my setup. After I did so, I had no access anymore.
Ecery attempt to flash again, gave me the same result inside the serial Monitor of Arduino IDE as shown above.
I do not understnd this behavior.

After I`ve edited the ircontoller.ino in the following way:
char static_ip[16] = "192.168.2.21";
char static_gw[16] = "192.168.2.1";
char static_sn[16] = "255.255.255.0";
it seems to be completely functionable for me.
That mean, it is running now.

Many thanks for your support.
and Greetings from Germany.
Thomas

eukiseioh, 2018-02-10:
👏👏👏👍

eukiseioh, 2018-02-10:
Thomas, i have the same NodeMCU board but i can’t send codes. I can learn perfectly.
Could you show me a pic of your config board pins conections and used components?

thommy6102, 2018-02-10:
I`ve just gave an answer and possible solution.
Just take a look to your request her on Github.

eukiseioh, 2018-02-10:
Thanks, i will try

Fabian234, 2018-02-12:
I set the configpin to ground but i have the same problem. Then i change the ip adresses to 
char static_ip[16] = "192.168.2.21";
char static_gw[16] = "192.168.2.1";
char static_sn[16] = "255.255.255.0";
but it still does not work. Does anyone have an idea what else could go wrong?

mdhiggins, 2018-02-12:
Make sure you're using lwIP 1.4 not the new default of 2.0
Don't change the IP in the blueprint that's a hack that probably won't work long term
If your config pin isn't causing the device to boot into AP mode then change your config pin to something else and set it to ground

mdhiggins, 2018-02-12:
@Fabian234 if you want help you need to define what 'not working' means

Fabian234, 2018-02-12:
the ESP8266 will display the same thing in the serial screen as above and will not start a WLAN access point

mdhiggins, 2018-02-12:
If its saying Config pin GPIO10 set to: 1 then the config pin isn't set to ground.

Its possible that on the V3 the default configpin can't be set to ground at boot so I would try a different pin and set that to ground

mdhiggins, 2018-02-15:
Sounds good

closed 2018-02-15 by mdhiggins

stresser1, 2021-07-05:
I'm using this board which I believe is a V3: https://www.pcboard.ca/nodemcu-v3?search=nodemcu

adjusting the Pins as per below works for me:

const uint16_t pinr1 = 14; // Receiving pin
const uint16_t pins1 = 4; // Transmitting preset 1
const uint16_t configpin = 12; // Reset Pin
const uint16_t pins2 = 5; // Transmitting preset 2
const uint16_t pins3 = 16; // Transmitting preset 3
const uint16_t pins4 = 13;