mdhiggins/ESP8266-HTTP-IR-Blaster #143: Externel IP Address

ldash94, 2018-02-03:
Hi i have successfully managed to get the IR blaster working. currently at the bottom of the webpage is says failed to retrieve external IP. 
I have put in my external IP and :9090 from my phone (on 4g) and successfully accessed the web page so i know the port forwarding etc is fine. 

Below is the comments from the bottom of the web page

38979ms uptime; EPOCH 38
Device secured with SHA256 authentication. Only commands sent and verified with Amazon Alexa and the IR Controller Skill will be processed
Error - unable to retrieve external IP address, this is likely due to improper network settings
Error - EPOCH time is inappropraitely low, likely connection to external time server has failed, check your network settings

Any suggestions what to try next? 

Thanks

mdhiggins, 2018-02-05:
It might be a DNS issue, are you using lwIP 1.4 or 2.0 when you compile and upload?

eukiseioh, 2018-02-10:
Yes, i solved it changing lwIP to 1.4 version in board serttings. Compile and upload again

ldash94, 2018-02-10:
I will try downgrading 

It works fine when I manually input the external ip

Thanks

closed 2018-02-16 by mdhiggins