mdhiggins/ESP8266-HTTP-IR-Blaster #144: How to change to lwIP 1.4 (Prebuilt)

katamara, 2018-02-05:
Sorry for the dumb question, but I cannot figure out how to do this step:
As of the ESP8266 Arduino Core 2.4.0 lwIP vesion 2.0 is default. This causes problems and crashing and it is recommended you use lwIP 1.4. Go to Tools > lwIP Variant > lwIP 1.4 (Prebuilt)

I have installed ESP8266 Arduino Core by Installing with Boards Manager 
Boards manager link: http://arduino.esp8266.com/stable/package_esp8266com_index.json

thanks!

eukiseioh, 2018-02-09:
In arduino app go to tools —> lwIP Variant —-> choose lwIP 1.4

I used lwIP 2.0 version with the same board

mdhiggins, 2018-02-09:
Yep, you just click those buttons its a setting in the Arduino IDE itself

Don't use lwIP 2.0 until they patch and fix it, I'll post an update whenever that happens

closed 2018-02-09 by mdhiggins

katamara, 2018-02-09:
In arduino app go to tools —> lwIP Variant —-> choose lwIP 1.4

I do not have this option under tools

eukiseioh, 2018-02-09:
Tools ——> into your board settings

eukiseioh, 2018-02-09:
Where you choose, frecuency, port, etc for select board

katamara, 2018-02-26:
Problem solved: I uninstalled The Arduino IDE and started all over again and this time lwIP Variant showed up under tools.
The issue was that this was not showing as an option the first time.

alphabeta1212, 2020-05-04:
Hi , can anyone tell me how can I change LWIP variant for MakeEspArduino Environment