mdhiggins/ESP8266-HTTP-IR-Blaster #145: Sent command

eukiseioh, 2018-02-09:
Hi, i can scan codes and are displayed on server web when i want send the command via web “http://xxx.xxx.xxx.xxx:port/msg?code=A90:SONY:12&pass=yourpass” say “code sent ok” but don’t work. Any problem with led emitter?
what’s happen? i used 100 ohm to resistor....

mdhiggins, 2018-02-09:
Probably a hardware problem
Verify using an IR capable camera (most front facing cameras on cell phones) that your LED is lighting up when you issue a command

If not then double check all your wiring (feel free to post a picture of your build and I can take a look)

eukiseioh, 2018-02-09:
Ok i will send a pic and check wired and led polarity too...

eukiseioh, 2018-02-09:
[image]

eukiseioh, 2018-02-09:
monitor output:
Connection received - MSG
Blasting off
Blocking incoming IR signals
FFD02F:nec:32
Transmission complete
Reenabling receiving
Turning off the LED to save power.

Success! code sent... but don't happens anything

the address opcion field is "0X0" when i learn any code...

mdhiggins, 2018-02-09:
The address behavior is normal. Only Panasonic codes use that. Wiring looks good so I would check resistors and LED polarity. Use a camera to see if the LED is lighting up as well. If no light then you’ll need to check all your hardware for a bad component

eukiseioh, 2018-02-09:
Mmmmm  if the LED is not lighting up i will try another resistor (120 ohm for example). I think the led polarity is good. 5,5v —-> resistor 100 ohm —-> + LED (longest pin).

thommy6102, 2018-02-10:
It seem that you have the V3 Revision of the NodeMCU.
I have had the same problem, and I had roundabout 1V only on Pin 30  Vin. I`ve used Pin 18 VV instead.
I`ve meassured 5V on this pin, and it works for me since several days.

eukiseioh, 2018-02-10:
@thommy6102 Works Fine!! i changed to Pin VV 
Thank you!

closed 2018-02-11 by mdhiggins