mdhiggins/ESP8266-HTTP-IR-Blaster #146: Channel Contoller does not work for me

thommy6102, 2018-02-10:
Hello again,

I`ve tested the IR Blaster with a Satellite Set Top Box.
On and off function and next and previous channel function works perfectly with Alexa.

But I do have problems with Channel Controller.
Maybe I`ve did some mistakes with the syntax.

I did it in that way:

Base URL: 
http://my_ip_address:667/json?

1 Button:
'281',type:'RC5',length:12

2 Button:
'282',type:'RC5',length:12
etc.
Is this correct?

If i use the URL`s directly like:
http://my_ip_address:667/msg?code=281:RC5:12
it works. But not after I edited my Device Setup.
Where I`m wrong?

Thanks in advance again for any support.
Thomas

mdhiggins, 2018-02-11:
Formatting is wrong
Channel controller uses the JSON formatting
Button 1 for example should look like:
`data:'281',type:'RC5',length:12`

mdhiggins, 2018-02-11:
Also no need to include the `?` in the JSON URL

thommy6102, 2018-02-11:
Thanks a lot. 
It`s working now. But not how expected.
I played a bit, and I detected that only the numbers 1,3,5,7,8 are working correctly.
The numbers 11,12,13, and 14 are ok also.
But not all the others.
Is there anything I can change or do?

Thanks again.
Thomas

thommy6102, 2018-02-11:
After I switched the language inside the Alexa settings page to german, everything until now works perfectly.
All numbers will be switched. That`s great.
So, I guess my above described problem have something to do with the lanuage skill.
Maybe german language is not completely supported?!?

mdhiggins, 2018-02-12:
Strange
Alexa translates the speech into numbers so if German isn't working its something on Amazon's end before it even reaches the skill

mdhiggins, 2018-02-12:
Looking through the German logs there are instances where Alexa is interpreting things incorrectly

Instead of sending '4' it occasionally sends things like 'kanal vier', so its an issue with Amazon and German probably

for a temporary workaround you could save 'kanal vier' in your channel list and map it to 4 and the script might recovery from these translation errors until Amazon improves things

closed 2018-02-13 by mdhiggins