mdhiggins/ESP8266-HTTP-IR-Blaster #147: NTP Error

Fabian234, 2018-02-13:
After i set the WLAN Configuration in the Browser in became the following error in the serial screen. Have anybody an idea what could go wrong?
[image]

Fabian234, 2018-02-13:
Where should I write the IP address of the router? In the program or in the browser in the wireless settings? Should i write the Gateway of the router in or the IP address?

mdhiggins, 2018-02-13:
In the browser in the wireless settings

Your device IP shoulder be similar to your router except the last segment of the IP

Your default gateway should match your router IP

And your subnet is probably 255.255.255.0

eukiseioh, 2018-02-13:
@mdhiggins i think you should upgrade tutorial and explain how to change ip settings when we are configure AP settings first time. By default set 10.xxx.xxx.xxx ip and i think anyone change it because don’t show how to do in the video tutorial.

Thank for all support and listen us

mdhiggins, 2018-02-13:
Its very basic level networking, like bare minimum so I would hope people attempting an Arduino DIY project would understand that. At some point if I have time I'll remake the video but my schedule is rough so it won't be for a while
The readme always has the most up to date installation instructions

closed 2018-02-13 by mdhiggins