mdhiggins/ESP8266-HTTP-IR-Blaster #148: Red Led Blinking and no Wifi

B0R1K, 2018-02-17:
any ideas?

Thank you.

P.S. Cant wait to get it running, good job...

B0R1K, 2018-02-17:
also i get garbage in serial monitor 

[image]

mdhiggins, 2018-02-18:
Gonna need a ton more information. What hardware are you using? What upload settings? Post your build pics, any setting changes?

B0R1K, 2018-02-18:
Thank you, here is the pic of my Build
[image]

B0R1K, 2018-02-18:
I bought this Board 
https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
and IR Receivers and Emitters 
https://www.amazon.com/gp/product/B00EFOQEUM/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

B0R1K, 2018-02-18:
Build Config
[image]

mdhiggins, 2018-02-19:
Update to ESP8266 Core 2.4.0

From: B0R1K
Sent: Sunday, February 18, 2018 3:27 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Red Led Blinking and no Wifi(#148)

Build Config

mdhiggins, 2018-02-19:
I would try removing the NodeMCU from the breadboard and uploading and see if that makes a difference
If it does it might be a short on the breadboard or a hardware issue, but everything looks wired correctly

You wanna change your lwIP Variant to v1.4 Prebuilt, not V2

You also can try changing your upload speed to 9600 and see if it makes a difference, though the Serial monitoring should be 115200

See if that helps

B0R1K, 2018-02-19:
Thank you for your continues help.

NodeMCU Behaves same when its on the BreadBoard and when its out. 

Also when I change the Upload Speed to 9600, it never completed and Fails with Error bellow at the same place, i tried multiple times:

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v1.4 Prebuilt, Disabled, None, 9600"

Archiving built core (caching) in: C:\Users\boris\AppData\Local\Temp\arduino_cache_247087\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_Prebuilt,Debug_Disabled,DebugLevel_None____,UploadSpeed_9600_85d5040a3eea385d60347e00175b8b88.a
Sketch uses 439976 bytes (42%) of program storage space. Maximum is 1044464 bytes.
Global variables use 50412 bytes (61%) of dynamic memory, leaving 31508 bytes for local variables. Maximum is 81920 bytes.
Uploading 444128 bytes from C:\Users\boris\AppData\Local\Temp\arduino_build_234826/IRController.ino.bin to flash at 0x00000000
................................................................................ [ 18% ]
................................................................................ [ 36% ]
................................................................................ [ 55% ]
.......................An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

crankyoldgit, 2018-02-19:
FYI, this is likely a faulty ESP8266 board. See https://github.com/esp8266/Arduino/issues/4005

B0R1K, 2018-02-19:
Thank you, any recommendation on new ESP8266 ?

mdhiggins, 2018-02-19:
https://www.amazon.com/Makerfocus-ESP8266-ESP-12E-Internet-Development/dp/B01IK9GEQG/ref=sr_1_3?ie=UTF8&qid=1519077308&sr=8-3&keywords=nodemcu

This is the one I use

B0R1K, 2018-02-19:
CP2102?

mdhiggins, 2018-02-19:
Yep

B0R1K, 2018-02-19:
cool, thank you.

closed 2018-03-05 by mdhiggins

B0R1K, 2018-03-06:
Thank you for your help, Just want to update, that after receiving new ESP, 
everything seems to be working... Will be playing with it for next few days...

Also ordered those Super Bright UI LED, as mine seems not to do anything... Checked with Phone Cam
I also did tested with regular LED just to see if blinks at all, and it did...