mdhiggins/ESP8266-HTTP-IR-Blaster #149: Error Compiling

inghaj, 2018-02-18:
Downloaded current version (18/2/17) and getting following error when compiling on Arduino IDE 1.8.5

"sketch\Alexa_IR_Controller_3_new_code.ino.cpp.o: In function `rokuCommand(String, String)':

... Alexa_IR_Controller_3_new_code/Alexa_IR_Controller_3_new_code.ino:234: undefined reference to `strtoull'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module)."

Would appreciate any help (was working fine a few versions ago!)

Thanks for your great work on this exciting project!
James

mdhiggins, 2018-02-19:
Update to Arduino ESP8266 Core 2.4.0

From: inghaj
Sent: Sunday, February 18, 2018 12:38 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] Error Compiling (#149)

Downloaded current version (18/2/17) and getting following error when compiling on Arduino IDE 1.8.5
"sketch\Alexa_IR_Controller_3_new_code.ino.cpp.o: In function `rokuCommand(String, String)':
... Alexa_IR_Controller_3_new_code/Alexa_IR_Controller_3_new_code.ino:234: undefined reference to `strtoull'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module)."
Would appreciate any help (was working fine a few versions ago!)
Thanks for your great work on this exciting project!
James

inghaj, 2018-02-20:
Great, problem solved. Thanks again for you great work on this  :-)
(I have a few questions (not issues) - Where is the best place to ask please?

closed 2018-02-20 by inghaj

mdhiggins, 2018-02-21:
You can just post them here