mdhiggins/ESP8266-HTTP-IR-Blaster #151: Trouble with JVC codes

DataBasye, 2018-02-21:
Great project!  I'm enjoying it.  I have my Blaster up and running.  Used one of Raptor's pcb boards for a nice compact unit.  (Now I'm looking for a nice project box to put it in--if anyone has a suggestion.) 

I'm just running into one problem.  Codes for my JVC receiver don't work.  Other entertainment components work (Vizio TV and Samsung DVD), but the JVC receiver just sits and stares at me. When I capture codes, they show up consistently as four character (length 16) JVC codes.  But the receiver won't react to any of them when I send.  I'm aiming right at the receiver from the same distance as the other components.  Alexa responds with an OK, and I see the led on the ESP8266 responding.  I also tried sending some raw codes.  Still no luck.  

Any suggestions on how to troubleshoot would be appreciated.

[image]

mdhiggins, 2018-02-21:
Can you post some example codes?

DataBasye, 2018-02-22:
The codes register consistently as 4 character JVC codes.  All of them start with "C5."

Here are a couple examples:

Here's what I get for the on/off button:
Data
C5E8
Type
JVC
Length
16
Address
0xa3
Raw
8438,4188,550,1556,552,1556,554,498,558,494,554,498,552,1554,556,498,554,1554,530,1576,556,1550,558,1550,556,494,556,1552,556,498,554,496,588,468,554

External IP MSG
http://...:56050/msg?code=C5E8:JVC:16

Here's what I get for the mute button:
Data
C538
Type
JVC
Length
16
Address
0xa3
Raw
8444,4184,558,1546,562,1544,558,516,540,518,536,516,508,1574,534,542,536,1496,608,520,508,544,534,1548,536,1572,558,1546,560,518,534,518,536,518,510

External IP MSG
http://...:56050/msg?code=C538:JVC:16

mdhiggins, 2018-02-22:
Try adding a pulse of 2 to 5 with a 50ms pdelay and see if that changes anything

DataBasye, 2018-02-22:
That did the trick.  Thank you!

closed 2018-02-22 by DataBasye