mdhiggins/ESP8266-HTTP-IR-Blaster #153: Question : Multiple IR Commands

inghaj, 2018-02-23:
Hi Michael
As your suggestion, I'm using the "issues" tab to to ask a couple of questions, even if they're not issues!

First one : Is it possible to send multiple IR signals from a single Alex command? 

As example, I'd like to create a device called "Cinema" - Then when I say "Alexa turn the cinema on", I'd like to issue several IR signals (turn amp on, turn projector on, turn tv off, switch Amp to HDMI 1 etc etc) ?

Many Thanks
James

mdhiggins, 2018-02-24:
You have a few options here

You can use Alexa's new built in "Routines" feature and just tie multiple device commands together, this works alright for simple power but if you want to get more complicated you can use the JSON commands to send multiple signals via the IRController

For example, I have my TV power on command also switch my surround sound input to TV and turn the roku to Plex

Here is what my URL for that looks like as an example

`json?plain=[{type:'NEC',data:'40BE629D',length:32},{type:'NEC',data:'FF13EC',length:32},{type:'roku',data:'keypress/home',ip:'10.0.1.33'},{type:delay,rdelay:1500},{type:'roku',data:'launch/13535',ip:'10.0.1.33'}]`

the readme goes over the syntax more, but essentially each item does the following

1. Turn on TV
2. Change surround sound input
3. Go home on the roku (closes anything running)
4. Wait 1500ms
5. Open plex on roku

All of this gets execute by Alexa when the turn on command is issued

The other option is to create a 'scene' using the scenecontroller with the command above just depending on your preference

closed 2018-02-24 by mdhiggins

inghaj, 2018-03-03:
Perfect, this is exactly what I was looking for. Thanks a lot - off to go and experiment now :-)