mdhiggins/ESP8266-HTTP-IR-Blaster #155: IR-Controller as Video Skill?

CSchoneweg, 2018-02-24:
Is it possible to get the IR-Controller skill as a Video Skill? At the moment it´s a Smart Home Skill. Unfortunately Smart Home Skills don´t support German except on/off commands. As described by amazon Video Skills are supporting German for the channel cotroller and the volume  controller too.

mdhiggins, 2018-02-24:
The video skills function completely differently and aren’t meant to have the same functionality that this implements, its not the correct option.
That being said it seems that other users aren’t have any issues using the German version of the skill, what issues are you having?

From: CSchoneweg
Sent: Saturday, February 24, 2018 6:04 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] IR-Controller as Video Skill?(#155)

Is it possible to get the IR-Controller skill as a Video Skill? At the moment it´s a Smart Home Skill. Unfortunately Smart Home Skills don´t support German except on/off commands. As described by amazon Video Skills are supporting German for the channel cotroller and the volume controller too.

CSchoneweg, 2018-02-25:
I can´t imagine that other german users have no problem. The answer from Alexa is "This device does not support this" if try to change the channel. If i change the language on one of my echos to english all works fine.

mdhiggins, 2018-02-25:
I can see the German log files for people using the channel controller, there’s been other support tickets for various pronunciation issues from people with German language Alexa but overall the functionality is enabled

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/146

From: CSchoneweg
Sent: Sunday, February 25, 2018 5:54 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] IR-Controller as Video Skill?(#155)

I can´t imagine that other german users have no problem. The answer from Alexa is "This device does not support this" if try to change the channel. If i change the language on one of my echos to english all works fine.

CSchoneweg, 2018-02-26:
The Channel Up Command http://my_IP:690/json?plain=[{'type':'NEC','data':'807FC03F','length':32}] and the Channel Down Command are working correct, the Power Controller and the Volume Controller too, but not changing a channel. 
My Base URL: http://my_IP:690/json
1 Button: data:'807F6A95',type:'NEC',length:32
2 Button: data:'807F5AA5',type:'NEC',length:32
Trying to change a channel gives the failure "This device does not support this". I´m not using a passcode but the amazon account ID.  So what is wrong?

mdhiggins, 2018-02-26:
How are you phrasing your Alexa request?

CSchoneweg, 2018-02-26:
In the meantime I tried many commands:
Alexa,
schalte zu kanal 2 auf tv (switch channel to 2 on tv)
wechsel zu kanal 2 auf tv (change channel to 2 on tv)
schalte zu 2 auf tv (switch to 2 on tv)
wechsel zu 2 auf tv (change to 2 on tv)
schalte tv zu kanal 2 (switch tv to channel 2)
wechsel tv zu kanal 2 (change tv to channel 2)
none of the above does work (tv does not support this, tv unterstützt das nicht)
But Alexa, Kanal rauf auf TV (channel up on tv) works!

One hour ago I changed the language to english U.K. to verify the commands in the IR Controller Setup.
With the english commands all is working correct, but ,of course, all my other skills do not.

mdhiggins, 2018-02-27:
Well they have not contacted me yet, I'll keep an eye out for the email

mdhiggins, 2018-02-27:
I would also like to clarify, on your support thread #146 you said things were working but certain channels were not behaving correctly

> Thanks a lot.
> It`s working now. But not how expected.
> I played a bit, and I detected that only the numbers 1,3,5,7,8 are working correctly.
> The numbers 11,12,13, and 14 are ok also.
> But not all the others.
> Is there anything I can change or do?

Is this still accurate? Perhaps ironically the communicate language barrier created a misunderstanding. What language setting was giving you the channel errors?

mdhiggins, 2018-02-27:
@CSchoneweg @thommy6102 

Can both of you email me your discovery JSON packet (go to bottom of the page on tehpsyc.pythonanywhere.com and email me the full JSON Object)

Additionally, can @thommy6102 provide the exact phrasing you're using for the channel controller and the error message you're getting. @CSchoneweg has already done this

mdhiggins, 2018-02-27:
According to the official Alexa documentation, the phrasing should be

`Alexa, wechsel auf Gerät den Kanal zu Arte`

I would try that, I would also maybe try a different device name than 'TV' since that is reserved for Spaces naming by Alexa and you can get conflicts with groups

CSchoneweg, 2018-02-27:
May be but the other functions like on/off and volume are working with the name "TV". However here is the Json Object

[
    {
        "cookie": {
            "powerOn": "http://31.18.220.233:690/json?plain=[{'type':'RC5','data':'84C','length':12,'rdelay':2000},{'type':'NEC','data':'807FFA05','length':32,'rdelay':1000}]&device=TV Kueche&state=1",
            "powerOff": "http://31.18.220.233:690/json?plain=[{'type':'RC5','data':'4C','length':12,'rdelay':2000},{'type':'NEC','data':'807FFA05','length':32,'rdelay':1000}]&device=TV Kueche&state=0",
            "volumeUp": "http://31.18.220.233:690/json?plain=[{'type':'RC5','data':'850','length':12,'pulse':5,'pdelay':75,'rdelay':400},{'type':'RC5','data':'50','length':12,'pulse':5,'pdelay':75,'rdelay':400}]",
            "volumeDown": "http://31.18.220.233:690/json?plain=[{'type':'RC5','data':'851','length':12,'pulse':5,'pdelay':75,'rdelay':400},{'type':'RC5','data':'51','length':12,'pulse':5,'pdelay':75,'rdelay':400}]",
            "mute": "http://31.18.220.233:690/json?plain=[{'type':'RC5','data':'4D','length':12}]",
            "baseURL": "http://31.18.220.233:690/json",
            "cup": "http://31.18.220.233:690/json?plain=[{'type':'NEC','data':'807FC03F','length':32}]",
            "cdown": "http://31.18.220.233:690/json?plain=[{'type':'NEC','data':'807FE01F','length':32}]",
            "c1": "data:'807F6A95',type:'NEC',length:32",
            "c2": "data:'807F5AA5',type:'NEC',length:32",
            "c3": "data:'807F7A85',type:'NEC',length:32",
            "c4": "data:'807FAA55',type:'NEC',length:32",
            "c5": "data:'807F9A65',type:'NEC',length:32",
            "c6": "data:'807FBA45',type:'NEC',length:32",
            "c7": "data:'807F2AD5',type:'NEC',length:32",
            "c8": "data:'807F1AE5',type:'NEC',length:32",
            "c9": "data:'807F3AC5',type:'NEC',length:32",
            "c0": "data:'807FD827',type:'NEC',length:32",
            "cent": "data:'807FA25D',type:'NEC',length:32"
        },
        "friendlyName": "TV",
        "endpointId": "Endpoint-001",
        "description": "Xoro Receiver und Medion TV via IR Controller",
        "manufacturerName": "MD Higgins",
        "displayCategories": [
            "TV"
        ],
        "capabilities": [
            {
                "interface": "Alexa.PowerController",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.StepSpeaker",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.ChannelController",
                "version": "1.0",
                "type": "AlexaInterface",
                "properties": {
                    "supported": [
                        {
                            "name": "channel"
                        }
                    ]
                }
            }
        ]
    },
    {
        "cookie": {
            "powerOn": "http://31.18.220.233:695/json?plain=[{'type':'SAMSUNG','data':818100FF','length':32}]",
            "powerOff": "http://31.18.220.233:695/json?plain=[{'type':'SAMSUNG','data':818100FF','length':32}]",
            "baseURL": "http://31.18.220.233:695/json",
            "c1": "data:'818140BF',type:'SAMSUNG',length32",
            "c2": "data:'818110EF',type:'SAMSUNG',length32",
            "c3": "data:'8181807F',type:'SAMSUNG',length32"
        },
        "friendlyName": "Sauger Schlafzimmer",
        "endpointId": "endpoint-002",
        "description": "Samsung Sauger SR-8845 via IR Controller",
        "manufacturerName": "MD Higgins",
        "displayCategories": [
            "OTHER"
        ],
        "capabilities": [
            {
                "interface": "Alexa.PowerController",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.ChannelController",
                "version": "1.0",
                "type": "AlexaInterface",
                "properties": {
                    "supported": [
                        {
                            "name": "channel"
                        }
                    ]
                }
            }
        ]
    }
]

thommy6102, 2018-02-28:
> 

>

thommy6102, 2018-02-28:
I played a bit, and I detected that only the numbers 1,3,5,7,8 are working correctly.
The numbers 11,12,13, and 14 are ok also.

Today, this is not reproducible. 
I`ve detected this behavior with german language setting. 
As I already wrote everything seems to be perfect afte my Alexa is switched to english.

thommy6102, 2018-02-28:
Here is my Json Object:

[
    {
        "cookie": {
            "powerOn": "http://my_IP_adress:667/msg?code=A8C:RC5:12",
            "powerOff": "http://my_IP_adress:667/msg?code=A8C:RC5:12",
            "mute": "http://my_IP_adress:667/msg?code=A8D:RC5:12",
            "baseURL": "http://my_IP_adress:667/json",
            "cup": "http://my_IP_adress:667/msg?code=AA0:RC5:12",
            "cdown": "http://my_IP_adress:667/msg?code=2A1:RC5:12",
            "c1": "data:'A81',type:'RC5',length:12",
            "c2": "data:'A82',type:'RC5',length:12",
            "c3": "data:'A83',type:'RC5',length:12",
            "c4": "data:'A84',type:'RC5',length:12",
            "c5": "data:'A85',type:'RC5',length:12",
            "c6": "data:'286',type:'RC5',length:12",
            "c7": "data:'A87',type:'RC5',length:12",
            "c8": "data:'A88',type:'RC5',length:12",
            "c9": "data:'A89',type:'RC5',length:12",
            "c0": "data:'A80',type:'RC5',length:12",
            "cent": "data:'1A97',type:'RC5',length:12"
        },
        "friendlyName": "Technisat",
        "endpointId": "Technisat-Receiver",
        "description": "Sat Receiver via IR Controller",
        "manufacturerName": "MD Higgins",
        "displayCategories": [
            "TV"
        ],
        "capabilities": [
            {
                "interface": "Alexa.PowerController",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.StepSpeaker",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.ChannelController",
                "version": "1.0",
                "type": "AlexaInterface",
                "properties": {
                    "supported": [
                        {
                            "name": "channel"
                        }
                    ]
                }
            }
        ]

thommy6102, 2018-02-28:
I`ve tested the following phrasing:
"Alexa schalte Technisat auf Kanal 1"
"Alexa wechsel Technisat auf Kanal 1"
"Alexa wechsel Technisat zuf Kanal 1"
"Alexa, wechsel auf Technisat zu Kanal zweihundert"

Error message by Alexa: Could not find activated Video Skills.......

All of them does not work.

mdhiggins, 2018-02-28:
So I guess Amazon changed something on the back end and things aren't being forwarded to the correct skill while previously it was working. I occassionally get the video skill error for input controller change if I saw the device name too quickly

Does the German version of Alexa support implicit targetting?
For example if you turn on a TV with alexa, for the next 15 minutes or so you can issue commands to that TV without specifying device name

I can for example use the following sequence of commands
Alexa, turn on the TV
Alexa, change channel to 52 (no need to specify TV or device name)

Do you get the same error when attempting that?

CSchoneweg, 2018-02-28:
yes, same error

CSchoneweg, 2018-02-28:
now something new happens. The Power Controller functions no longer. Answer from Alexa is "OK", Wemos blinks but my TV does nothing and my seperate IR Receiver receives nothing. I did not change any command. What´s going on?

mdhiggins, 2018-02-28:
I haven't changed anything on the back end, and if the LED on the Wemos lights up then that means the command was received successfully from Alexa so you need to check your hardware and LED and verify you didn't change any of your remote code settings

If you reflashed your firmware and had previously changed your pinouts that's another possible area of error

CSchoneweg, 2018-02-28:
That´s absolut grazy, I haven't changed anything. The hardware must be ok too because the volume controller works!

CSchoneweg, 2018-02-28:
never mind, on/off function is back. And thanks for the hint with the implicit targetting. The commands are much easier now.

thommy6102, 2018-03-01:
I`ve detected the same result, concerning the implicit targeting.
After a while, there is no need to use a specified name.

mdhiggins, 2018-03-01:
That’s a feature that was added by amazon many months ago. Does channel controller work with implicit targeting if you try without a device name after turning the TV on?

CSchoneweg, 2018-03-01:
No

thommy6102, 2018-03-07:
I`m going to contact the Amazon Team again.

thommy6102, 2018-03-10:
I`ve contacted Amazon Germany via telephone this morning.
They are going to solve the problem.
Let`s see what we get.

Thomas

thommy6102, 2018-03-17:
Until now, I didn`t get feedback by Amazon.
But I figured the following out:
I have defined 2 devices inside my IR Controller Setup. 
1. A Set Top Box for Satellite Receiption.
2. A TV.
If I say:" Alexa switch on channel xx" in german: "Alexa schalte auf Kanal xx" then in works for the following
channel numbers.
1,8,9,11,12,14,19,20,22,29. (Channels above 30 were not tested)
Any other channel numbers do not work. This is valid for both devices. Set Top Box and TV Set.
I`ve checked it several times, and the behavior is still the same.
For my naiv opinion, there must be a difference to the working and non working channels.
But  where it is?

mdhiggins, 2018-03-17:
I actually explained what was happening in #146 and why some channels were working and some weren’t

Looking through the German logs there are instances where Alexa is interpreting things incorrectly

Instead of sending '4' it occasionally sends things like 'kanal vier', so its an issue with Amazon and German probably

for a temporary workaround you could save 'kanal vier' in your channel list and map it to 4 and the script might recovery from these translation errors until Amazon improves things

Unfortunately all this processing happens before the command reaches the skill and is something Amazon will have to fix

thommy6102, 2018-03-19:
Last Saturday I`ve got feedback by Amazon.
Here`s the short translation what they wrote:
"Since this is a third party skill, I ask you to contact the manufacturer Michael Higgins directly.
I'm sorry that I can not give you any further information and hope for your understanding."
A pity, I had hoped for a little more.

CSchoneweg, 2018-03-28:
It seems as amazon did something. Suddenly the channel controller works fine if the channels are called by number. Calling the channels by name still does not work.

thommy6102, 2018-03-28:
Realy?
I can`t believe that. When did you discover it? 
I`m lvery looking forward to check it later on, after work is finnished.
Thanks for info.

CSchoneweg, 2018-03-28:
Every time i use my kitchen tv i´m trying that and today in the morning it worked!!

mdhiggins, 2018-03-28:
That’s great news. When you say the channel names aren’t working what issues are you having? And did you configure the channels?

CSchoneweg, 2018-03-28:
Yes I configured the channels, but "... is not supported on this device".

mdhiggins, 2018-03-29:
Hm this must just be Amazon still actively working on true channel controller support
Their documentation indicates named channels should work but obviously that may not be the case
Since this is more of an amazon issue and unrelated to the skill I'm gonna close this
Feel free to continue discussion or reopen if you have a different issue

closed 2018-03-29 by mdhiggins