mdhiggins/ESP8266-HTTP-IR-Blaster #156: Am I using the wrong board?

katamara, 2018-02-26:
I am having reboot issues (rst cause:4, boot mode:(3,6))

I am wondering if I may possibly be using the wrong board.

This is the board I am using:

https://www.aliexpress.com/item/NodeMcu-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-module/32668278518.html

IR Receiver:
https://www.aliexpress.com/item/10pcs-Lot-Universal-IR-Infrared-Receiver-Sensor-Module-1838-TL1838-VS1838B-38Khz-Diode/32507491875.html

2N222:
https://www.aliexpress.com/item/100pcs-free-shipping-2N2222-TO-92-40V-0-8A-NPN-switching-transistors-new-original/32506958631.html

IR LED:
https://www.aliexpress.com/item/50Pcs-5mm-940nm-IR-infrared-LED-Lamp/32224584146.html

mdhiggins, 2018-02-27:
Nope, that's the right board

The error you're getting is usually from having something connected to the wrong pin or having installed a library that wasn't meant for the ESP8266 (perhaps something for Arduino without ESP8266 support). Double check all your installed libraries and try booting the board without it being connected to anything except USB then slowly add back hardware

closed 2018-03-05 by mdhiggins

katamara, 2018-03-05:
I got rid of the reboot issues by changing the board in the Arduini IDE from NodeMCU v0.9 to NodeMCU v1.0