mdhiggins/ESP8266-HTTP-IR-Blaster #158: *noob* External IP error (do I need a static IP for this project to work?)

maxjosh, 2018-03-05:
Hi
I have reached the config screen of IR Blaster. I have an ADSL modem linked to my wifi router with internal IP 192.168.0.1, which I put as the static gateway address. Static IP of IR device- 192.168.0.67
The serial monitor shows the following issues-

_Error retrieving external IP HTTP Code: -1 connection refused External IP
address request took 9 ms Warning, user_id appears to be in the wrong
format, security check will most likely fail. Should start with
amzn1.account.*** Epoch time from timeServer is unexpectedly old, probably
failed connection to the time server. 
Check your network settings
14400_

If this is incorrect than what static IP should I put?
I understand the part about the user_id format, but unsure why the external IP is not connecting.
Thanks for reading..

mdhiggins, 2018-03-05:
There's a bug with lwIP version 2.0 and DNS, and lwIP 2.0 was made the default selection with ES8266 2.4.0
When you compile please choose lwIP version 1.4 and this usually fixes this error assuming your other networking settings were entered appropriately

`Tools > lwIP Variant > lwIP 1.4 (Prebuilt)`

chrislawson, 2018-03-15:
Changing the IwIP Variant fixed this issue for me as well on a Win10 laptop running Arduino 1.8.4. 

I wanted to add that 'lwIP 1.4 (Prebuilt)' was not on the list of choices and that the only one's listed for me are 'v2 Lower Memory', 'v2 Higher Bandwidth', 'v1.4 Higher Bandwidth' and 'v1.4 Compile from source'.

I'd originally tried 'v1.4 Compile from source' but my system was missing the 'make' executable. I then switched to 'v1.4 Higher Bandwidth' and that worked.

mikep2018, 2018-03-19:
Michael great work.  When I use lwIP 2.0 I can't setup the external IP.  When I use 1.4 I can see the external IP but then I can't access the IR page.  Is there a way to hard code my external IP into the source?

mdhiggins, 2018-03-19:
So the external IP showing isn’t actually a problem in itself it’s just a symptom of networking not actually functioning properly. Usually a DNS issue. The external IP isn’t actually used or stored just displayed for user reference but the failure means the network request didn’t work. 

What error are you getting on 1.4 that’s preventing things from working? Cause that should work

mikep2018, 2018-03-19:
Michael.  When using 1.4 the dongle.local page just hangs and doesn't load.  When I use 2.0 it loads and lets me read IR codes.

mdhiggins, 2018-03-20:
What board do you have set and what options specifically do you have for lwIP ? (prebuilt vs higher bandwidth etc)

Additionally, when using lwIP 1.4 and being unable to access the local page, have you tried using the IP instead of the mDNS url? Can you also post the full serial output from boot to your access attempt?

timo127, 2018-03-25:
I have exactly the same problem.
Esp8266 connects to my network with IwIP1.4 according to the monitor, but I cannot access it. With 2.0 the IR controller appears.

mdhiggins, 2018-03-25:
Can you ping it? Does IP work vs mDNS? Can you post your serial output from startup and then while you try to access?

eightonegulf, 2018-03-26:
I have the same issue, as described above..
When using lwIP 1.4 Higher Bandwidth the Esp8266 is not pingable from a local machine. It does manage to connect to dyndns.org to fetch it's external IP adres. 

After that it seems to hang. With the console open it does not recognize any IR codes.
Below is the serial output if you're interested, but there is no useful information related to this issue

```
Should save config
WiFi connected! User chose hostname 'irblaster' passcode '*****' and port '80'
 config...
{"hostname":"irblaster","passcode":"*****","port_str":"80","user_id":"","ip":"192.168.2.80","gw":"192.168.2.254","sn":"255.255.255.0"}
Writing config file
Config written successfully
WiFi configuration complete
Local IP: 192.168.2.80
URL to send commands: http://irblaster.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to irblaster.local:80
HTTP Server started on port 80
External IP: *********
External IP address request took 99 ms
Ready to send and receive IR signals
Turning off the LED to save power.

```

Running lwIP 2.0 does recognize IR codes, but hangs after a while (as described in the readme), and as described in this thread refuses to connect to the internet

Hope this is any help :)

mdhiggins, 2018-03-27:
`using lwIP 1.4 Higher Bandwidth`

That's a clue right there
The 'higher bandwidth' option is for a different board which implies to me that you might have the wrong board set on board manager

Try setting board to `NodeMCU 1.0 (ESP-12E Model)` and you want to make sure you're using esp8266 version 2.4.0 in board manager as 2.4.1 has a memory leak that has not be fixed yet that causes freezing.

mdhiggins, 2018-03-27:
@eightonegulf And I just want to confirm that the local IP address of your router is 192.168.2.254, is that correct?

eightonegulf, 2018-03-27:
192.168.2.254 is indeed the IP address of my router. 

I had 2.4.1 installed and with NodeMCU 1.0 (ESP-12E Module) selected it only gave me the lwIP 1.4 Higher Bandwidth and lwIP 1.4 Build from source, and the 2.0 options. 
I downgraded to 2.4.0 and it offered me the 1.4 PreBuild option.

Running with these options seems to be working stable.

So downgrading from esp8266 2.4.1 to 2.4.0 seems to have fixed my issue.
Thank you mdhiggins!

mikep2018, 2018-03-28:
Michael.  Sorry I haven't replied sooner (busy).  I just read over the previous posts and that was my issue too.  I downgraded to 2.4.0 an it works.  Thanks for your help!!  Keep up the great work!!!!

mdhiggins, 2018-03-29:
Good news

closed 2018-03-29 by mdhiggins