mdhiggins/ESP8266-HTTP-IR-Blaster #160: Is there way to update Wifi Settings, without reseting ?

B0R1K, 2018-03-14:
Also is there way to safe External port, if you doing port forwarding...  
lets say from from 8080 external to 80 internal...

Thank you,

jebbett, 2018-03-15:
What do you mean by resetting? The easiest option is to add a button or short the config pin to ground, do that as you press the reset button or add power..

For port forwarding, what you are saying is possible assuming your router supports it, but alternatively if your router doesn't then  just set the device port to something random.. I would probably pick something more cryptic than 8080 or 81 as both are frequently used.. but a bit of port sniffing would still discover the device.. I wouldn't call opening it up safe anyway, but relatively low risk..

Also if you have just added the ESP8266 board library, V1.4.1 has a memory leak and you will likely get issues, 1.4.0 works fine.. just to save you some headaches :)

B0R1K, 2018-03-16:
my point is that without saving external port somewhere in config, External URL get the wrong port...

Also i was asking if there a way to navigate to Wifi Config page without pressing a button, maybe there is hidden url or something... in case i what see what i entered... etc...

Thank you for Reply, i been playing with project and so far really like it...

jebbett, 2018-03-16:
The external port will be the same as the internal port if you just do a standard port forward, if you want to re-map this to something else externally on your router then you can.. when copying URLs from the web front end you can then just amend the :80 to :8080 or whatever you need..that's the easiest option here. It could be updated in code, but there are reasons to specify the internal port in the code as the web server needs to operate on that port, so it's mainly there for that purpose.. but the external port makes no difference other than displaying the link (I believe, I don't think this is used by Amazon?).. if you are keen to update the code you could do this by adding duplicates of the below fields thorough out the code with a prefix of ext then update the appropriate entries at about line 963 to refer to the extport instead of port

port
port_str

I had a play last night to see if I could get to the config page without pressing the button, I had no luck either.. it would be a good idea.. except that someone without physical access to the device could then in theory change the settings (a risk when you open to an external port).. This may be the reason it's not been opened, or could just be a code or memory limitation..

I've only been playing with this for a few months, and I now have a couple of these around the house linked to SmartThings.. I think this is an excellent project and quite simple even for a novice!

devax, 2018-03-18:
> Also if you have just added the ESP8266 board library, V1.4.1 has a memory leak and you will likely get issues, 1.4.0 works fine.. just to save you some headaches :)
Could you clarify which library you mean exactly? I checked all libraries mentioned in the README but could not find one with version number close to 1.4.0 / 1.4.1.

jebbett, 2018-03-18:
Yes sorry for the confusion!

closed 2018-03-19 by mdhiggins