mdhiggins/ESP8266-HTTP-IR-Blaster #161: undefined reference to `strtoull'

error0327, 2018-03-14:
Hi , 
i installed all libraries , and download the Cryptosuite to the lib dir , and win compiling getting the following error :

Archiving built core (caching) in: C:\Users\obadaa\AppData\Local\Temp\arduino_cache_588465\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M_2c866e82cf51a3a3653ab1299af6ca1c.a
sketch\ESP8266-HTTP-IR-Blaster.ino.cpp.o: In function `rokuCommand(String, String)':

C:\Users\obadaa\Desktop\Ardinu\ESP8266-HTTP-IR-Blaster/ESP8266-HTTP-IR-Blaster.ino:234: undefined reference to `strtoull'

C:\Users\obadaa\Desktop\Ardinu\ESP8266-HTTP-IR-Blaster/ESP8266-HTTP-IR-Blaster.ino:234: undefined reference to `strtoull'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board WeMos D1 R2 & mini.

mdhiggins, 2018-03-14:
What version of the ESP8266 Arduino Core are you using?
Need to use the latest (2.4.0) and it seems like you’re using an older version

From: Obada abu alhiga
Sent: Wednesday, March 14, 2018 2:17 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] undefined reference to `strtoull'(#161)

Hi ,
i installed all libraries , and download the Cryptosuite to the lib dir , and win compiling getting the following error :
Archiving built core (caching) in: C:\Users\obadaa\AppData\Local\Temp\arduino_cache_588465\core\core_esp8266_esp8266_d1_mini_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M_2c866e82cf51a3a3653ab1299af6ca1c.a
sketch\ESP8266-HTTP-IR-Blaster.ino.cpp.o: In function `rokuCommand(String, String)':
C:\Users\obadaa\Desktop\Ardinu\ESP8266-HTTP-IR-Blaster/ESP8266-HTTP-IR-Blaster.ino:234: undefined reference to `strtoull'
C:\Users\obadaa\Desktop\Ardinu\ESP8266-HTTP-IR-Blaster/ESP8266-HTTP-IR-Blaster.ino:234: undefined reference to `strtoull'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board WeMos D1 R2 & mini.

error0327, 2018-03-14:
i don't have such version , i am using Adafruit lib , which one you have installed ?

jebbett, 2018-03-15:
You need to follow the instructions and add the ESP8266 board library in the settings page..

error0327, 2018-03-15:
What was not clear of it's already installed but not working ?! , i have Adafruit IO Arduino v 2.7.1 installed which include all the ESP8266 lib's , but it's look like another lib needed , in the instructions there is info regarding lib source .
i am using the following source : http://arduino.esp8266.com/stable/package_esp8266com_index.json

i hope it's clear and finally someone can share the full version name and source .
because searching for :

Install the following libraries from the Arduino IDE Library Manager: ESP8266WebServer ESP8266WiFi ArduinoJson WiFiManager EasyNTPClient IRremoteESP8266 as well as Cryptosuite which is not on the IDE

is not working !

kokarn, 2018-03-16:
This is actually a bug that I forgot to PR.
There's still one reference to `strtoull` in the code here https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/master/IRController/IRController.ino#L1311

Seems the fix from https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/commit/b4e0f15215382f6a8eb6da031abf487af7bf1147 got reverted here https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/commit/8fb3e70a67d658e13a368460dd763394ad8c947d

mdhiggins, 2018-03-16:
Lots of misinformation here
@kokarn is incorrect strtoull is not a typo and is different in functionality than strtoul

strtoull creates an unsigned long long (as opposed to unsigned long) and is needed for decoding things with higher memory usage

It was added in ESP8266 Arduino Core 2.4.0 and you need to update to this to support it
You install this via board manager in Arduino IDE

[image]

kokarn, 2018-03-16:
Glad I was wrong about that then. Sorry for diluting the issue

mdhiggins, 2018-03-16:
No worries I didn’t know that initially either and reverted after doing some reading. Just wanted to clarify and set things straight

error0327, 2018-03-16:
Thanks everyone , i do remove all my resources  , clean AppData , reinstall Arduino IDE , reinstall it all , and now it compiled with no issues.

mdhiggins, 2018-03-16:
Great glad it’s woeking

closed 2018-03-16 by mdhiggins