mdhiggins/ESP8266-HTTP-IR-Blaster #163: Assistance with lockup after boot?

chrislawson, 2018-03-16:
This is such a great project! Thank you for making it available. 

I followed the steps in the readme to set everything up on a NodeMCU and did need to use the 1.4 instead of 2.0. I've configured wifi and It boots just fine after that however, it seems to freeze about 15 seconds after that. I have just enough time to open the root web page but it stops responding after that. 

Can you offer any ideas one what I can check? Here is the output of the serial monitor with ext ip and pwd redacted.

ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"irblast","passcode":"****","port_str":"80","user_id":"","ip":"192.168.1.71","gw":"192.168.1.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.71
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.71
WiFi connected! User chose hostname 'irblast' passcode '******' and port '80'
WiFi configuration complete
Local IP: 192.168.1.71
URL to send commands: http://irblast.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to irblast.local:80
HTTP Server started on port 80
External IP: xxx.xxx.xxx.xxx
External IP address request took 376 ms
Ready to send and receive IR signals
Turning off the LED to save power.
Connection received

mdhiggins, 2018-03-16:
Hm that’s a tough one it looks like everything is working as it should. What exactly freezes and after freezing does it crash or do you get any reports in the serial log or just what’s posted?

If you remove all connections from your board (disconnect from breadboard) and try again does that make any difference?

chrislawson, 2018-03-16:
Thanks for the quick response. I am currently testing with board only with nothing connected and no breadboard.

I am not sure how else to describe the 'freeze'. The board no longer responds to HTTP requests nor does it respond to IR signals when I had the receiver hooked up. Nothing else appears on the serial monitor after this point. The only way to get it to respond again is to press the reset button but then it just freezes again.

I tried another quick test by rebooting and then refreshing my browser as quickly as I could. I was able to get 3 requests in before it died. Here's all that is at the end of the log.

<SNIP>
Turning off the LED to save power.
Connection received
Connection received
Connection received

I am sure that the issue is related to my setup or something that I did, not necessarily a bug with your code. I'm a noob to this and was hoping for any guidance to help me troubleshoot and resolve.

FCustoms, 2018-03-17:
I have the same problemma, the receive-transmits about 5 times and freezes. only the reboot works.

mdhiggins, 2018-03-17:
Hm I have about 4 primary devices and multiple test devices and most of them have uptime measured in the months without rebooting. What hardware are you guys using that’s definitely not a standard issue

FCustoms, 2018-03-17:
I'm using LoLin's new NodeMcu V3 (serial ch340g). And the reset of the pins does not work, maybe this problem? 
....I'll try to disable today the transfer serial, maybe can help.

mdhiggins, 2018-03-17:
Yeah I haven’t tested V3 all my boards are V2 so it might be a short on a pin. You can play around with the default pin outs and maybe try moving the receive and send pins around too. If you notice it fails after receiving a signal then maybe the receive pin is an issue and needs to be changed.

chrislawson, 2018-03-17:
I tried the full setup again from my mac instead win laptop and get the same results though, I was able to use the IwIP v2 instead of the v1. Upon boot the board responded for about a full minute then quit.

I am not sure what version my board is but I think at this point I need to order a new one and try again.

Feel free to close this. Thanks!!

devax, 2018-03-17:
Hey guys, I have a similar problem. I'm also using the so called "V3" variant of this board from LoLin. Everything seemed to work fine, but when I send multiple commands in a row, like pulse=3, the device will freeze and I will have to press RST button. By freeze I mean it will not respond to ping anymore and is not accessible. I have noticed, when I send a single IR command, the command gets executed, but one ping packet will be highly delayed (like 200ms compared to 3ms normally).
I wonder if it could be an "out of power" issue or something like that?
While assembling the board I found out that apparently you cannot use VIN with these boards; instead I used VV for 5V power supply.
The reason I hope to find a fix for that is because first tests looked good so I have just finished soldering everything together what took hours (guess I'm not too good in soldering) so it would be a pain to do it again with another board. Hope someone has a hint for me! Thanks!

devax, 2018-03-18:
I was able to resolve my freezing issues by downgrading the esp8266 board library in Arduino board manager from version 2.4.1 to version 2.4.0. This has been mentioned in issue #160 and this worked for me.

jebbett, 2018-03-18:
It’s a memory leak in lwip in 2.4.1, there is a bug raised in the project, hopefully fixed in 2.4.2, but as mentioned above 2.4.0 works fine..

FCustoms, 2018-03-19:
I confirm! changed version to 2.4.0 and it all worked

chrislawson, 2018-03-19:
I have had limited success with changing to 2.4.0.  I no longer experience freezes but only if I choose the the IwIP v2 Lower Memory or v2 Higher Bandwidth. Using v2 prevents retrieval of my external IP when I bring up web interface. It reports this message:

`Error - unable to retrieve external IP address, this is likely due to improper network settings`
`Error - EPOCH time is inappropraitely low, likely connection to external time server has failed, check your network settings`

If I use the IwIP v1.4 in combination with 2.4.0, I still get the lockup.

jebbett, 2018-03-19:
@chrislawson In 2.4.0 I don't get those options.. I only get the below, so may not be removed correctly? 

- v2 Prebuilt (MSS=536)
- v2 Prebuilt (MSS=1460, unstable)
- v1.4 Prebuilt **<<< I'm using this**
- v1.4 Open Source

I'm not using external IP so can't add much more, although the guide does say to use lwip 1.4 as 2.0 causes issues.

chrislawson, 2018-03-19:
@jebbett you are correct. I looked at the list of boards and found there to be TWO ESP8266 Modules sections. I selected the correct board from the other list and found the v1.4 Prebuilt that was described.

Everything is now working exactly as expected. Very happy!

mdhiggins, 2018-03-19:
Glad its all working

closed 2018-03-19 by mdhiggins