mdhiggins/ESP8266-HTTP-IR-Blaster #164: https

Fabian234, 2018-03-17:
Can I do the transfer with https even if I do not use Alexa?

mdhiggins, 2018-03-17:
The ESP8266 does not have enough memory to do HTTPS and this device doesn’t use it 

The SHA256 HMAC authentication can be done without Alexa you’ll just have to write a program that goes through the same encryption process and generates the tokens required. Could be easily made in python or node JS just go through the same sequence that the blueprint does

Fabian234, 2018-03-17:
How do I do that in nodejs exactly?

mdhiggins, 2018-03-17:
I mean you’ll have to do some reading and learn some crypto basics I probably wont have time to write it for you. It’s an open source project. 

You just need to add components to the HMAC in the same order to generate the key

If I get around to it I’ll try but no promises

closed 2018-03-19 by mdhiggins