mdhiggins/ESP8266-HTTP-IR-Blaster #165: Mqtt support coz port forwarding is not available

bhasinshubham, 2018-03-17:
My ISP does not allow port forwarding(many must be facing this 🤔) . Can someone help me set this up as a Mqtt client? There are many free mqqt broker services online and this would definitely make the project more appealing (would work on mobile connections, etc, etc) and easy to install/use. Thanks in advance.

mdhiggins, 2018-03-17:
MQTT would require a complete rewrite of both the Alexa skill and the Blueprint and reduce some of the functionality for complex combination commands

I would look at the MQTT sample provided by the IRRemoteESP8266 library if you’re looking to build a solution

https://github.com/markszabo/IRremoteESP8266/tree/master/examples/IRMQTTServer

closed 2018-03-17 by mdhiggins

bhasinshubham, 2018-03-17:
Thanks a ton for your prompt and helpful response... I will definitely check out the sample.