mdhiggins/ESP8266-HTTP-IR-Blaster #167: IR LED is not sending the signal

kiranreddyr, 2018-04-05:
i used the both the circuits i'm just receiving the signal but when i try to send the signal IR LED is not Blasting any signal      below i have attached the circuit connections i have made....
[image]
[image]

mdhiggins, 2018-04-05:
A little hard to tell from the pictures but it looks like everything is wired correctly. Make sure you don’t have your LED polarity reversed as that will prevent signal from being sent. 

Next question is what command are you trying to send? Does everything else appear to be working without issue? No problems from the code? Post some logs of your attempts. 

You can also check and see if the LED is lighting up using a camera without an IR filter (most cel phone front facing cameras). 

Can also swap the LED with one from visible spectrum and see if it lights up

kiranreddyr, 2018-04-07:
i think the problem is with the code so can i get link for default.bas code

mdhiggins, 2018-04-07:
Default.bas is very old and works with none of the new features, does not work with Alexa
The problem is most certainly not with the code. The script and devices have hundreds of active daily users that aren't having issue. Its most likely a hardware issue. You can try connecting the LED directly to the data pin and eliminating the transistor
You can also try removing your resistor to the LED and see if that makes a difference
What resistors are you using? and have you confirmed you have the right ones?
You can also try taking a different LED (visible light, not IR) and see if anything is emitted

mdhiggins, 2018-04-07:
I guess I should also verify, what URL are you using to send the commands? Are you doing it through Alexa or browser? What remote code exactly are you sending? What does the website say when you send the command, is there an error? and finally post the serial output logs when you're attempting to send codes

kiranreddyr, 2018-04-09:
the code which i have to send is attached in the picture below....
[image]

To send the code i have to  press the Button or i have to go to browser and type this URL 192.168.4.1/msg?code=1

kiranreddyr, 2018-04-09:
If i hit this URL in Browser 192.168.4.1/msg?code=1   i'm just getting response as shown in below picture
[image]

mdhiggins, 2018-04-09:
Version 1 of the code is no longer supported
It uses a very old version of the IR Remote Library, has no security, and does not use Arduino
You should upgrade to the latest version

From: kiranreddyr
Sent: Monday, April 9, 2018 2:31 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] IR LED is not sending thesignal (#167)

If i hit this URL in Browser 192.168.4.1/msg?code=1 i'm just getting response as shown in below picture

kiranreddyr, 2018-04-11:
so how can i upgrade to the latest version?is it by changing the code in Edit field to the the IRcontroller.ino code what u have given?

mdhiggins, 2018-04-11:
No, the ESP8266 boards actually support running the Arduino firmware and code

What you'll need to do is download and set up the Arduino IDE and go through the readme for the setup process for which dependencies you need. If you've never used Arduino or flashed a board I would look up some Youtube videos for people doing it with the NodeMCU. The .ino file is an Arduino file that you'll use to reflash the board

Once you get that working the Youtube video in the readme will be a good place to start

https://www.arduino.cc/en/Main/Software
https://youtu.be/JysKXOdrOlM

kiranreddyr, 2018-04-11:
i have already flashed NODEMCU so now is it possible to use the same board
or i have to use different board and do from first?

*Thanks & Regards*

*Kiran R*
*9036288730*

kiranreddyr, 2018-04-11:
HI Micheal,
is it possible for u to give some time so that we can have a chat...
You can say time when ever you are free so ill come online that time
pls i need your help Micheal..

*Thanks & Regards*

*Kiran R*
*9036288730*

kiranreddyr, 2018-04-11:
if its possible then we can share screen through Hangouts or Team Viewer

*Thanks & Regards*

*Kiran R*
*9036288730*

mdhiggins, 2018-04-11:
Nah you can reflash as many times as you want

mdhiggins, 2018-04-11:
Sorry but I’m very busy and won’t have time to walk you through the process individually. I can answer specific questions they come up by email but that’s about the best I can do

kiranreddyr, 2018-04-12:
ok.. i already flashed that NODEMCU board and i'm able tho access IP address also 

But now when i try to program it with Arduino IDE im not able get any COM port to program it.... 

i used NODEMCU flasher in beginning to flash this NODEMCU ESP8266

and now when i try to change the code version as u say im not getting any COM port

mdhiggins, 2018-04-12:
I can’t teach you the basics of using the Arduino software. There are thousands of projects and hundreds of YouTube videos that can walk you through it. This is not specific to my project this is a general Arduino question. Open source DIY projects require you to familiarize yourself with the background information before getting started

I will say there have been lots of bug fixes with the Panasonic codes in the later versions so you might have luck getting them working if you put the time in. I’m happy to try and field specific questions when you get further but you gotta put in the effort too.

closed 2018-05-03 by mdhiggins