mdhiggins/ESP8266-HTTP-IR-Blaster #170: Error retrieving external IP

muditgoel92, 2018-04-22:
Hi, would you be able to help with the below?

Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 5002 ms
No errors detected with security configuration
Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
14400
Invalid EPOCH time, security checks may fail if unable to sync with NTP server

mdhiggins, 2018-04-22:
Check your network settings 

Also check and make sure you’re using ESP8266 core 2.4.0 with lwIP prebuilt v1.4

muditgoel92, 2018-04-22:
Yes, I am using 2.4.0 + 1.4.
Checked network setting. All set to automatic NTP sync.

Any further suggestions?

muditgoel92, 2018-04-22:
after a bit of network trouble shooting, I was able to resolve this. Thanks.

closed 2018-04-22 by muditgoel92

slimlesaurien, 2020-04-26:
Bonjours

slimlesaurien, 2020-04-26:
Bonjours muditgoel j'ai le même problème 
Qu'à tu fait pour le résoudre merci d'avance

riskyrun, 2020-06-07:
> after a bit of network trouble shooting, I was able to resolve this. Thanks.

Noob here.What kind of troubleshooting?