mdhiggins/ESP8266-HTTP-IR-Blaster #171: Two isses: External IP access and keeping the device on

muditgoel92, 2018-04-28:
I am facing two issues:
- While accessing the external IP, it is asking for user ID and password. What ID / password combo is this? I have used my amazon ID / password, Amzn1 ID / password that gave during set-up phase here, host name / password combos, none seem to work

- I can see that the code is compiled and I am able to open the controller from my local IP address. However, I have to everytime press the RST button on the ESP8266, everytime I refresh the page. Also, On the source log, it is able to receive the IR code only once. Not able to recevie it again and it says "turning off LED to save power". After that it dosent receive ay command nor am I am able to refresh the local IP page.

Overall, I think this is the reason the transmitter is also not transmitting it to my TV and Alexa commands are not working.

Your help would be highly appreciated.
Regards,
Mudit

muditgoel92, 2018-05-01:
Hi, checking if someone can assist me here?

muditgoel92, 2018-05-03:
[image]
[image]
Please find attached. Thanks again for your help.

mdhiggins, 2018-05-03:
Yeah that login is not from the device
I think your ISP is intercepting the signal on that port
You can try using a different port instead of 80, 80 is often reserved

closed 2018-05-12 by mdhiggins