mdhiggins/ESP8266-HTTP-IR-Blaster #172: Trouble with AC IR replication

sorinGrecu, 2018-04-28:
Hi, first of all, awesome tool!
I hooked everything up, used a Keyes IR blaster kit instead of a led with resistors and transistors. It works for a SONY Tv and Yamaha AV Receiver with no problems.
The issue is when I try it for a no-name AC unit. The "brand' is Americool, I guess it's just a low cost chinese thing with various small companies branding them, had various no name AC units where I lived that looked the same but with different branding.
Anyway, I receive the code but when trying to send it, even tho the ir led lights up (can be seen with a phone camera) the AC does not turn on/off. Tried multiple times but to no avail.
This would be the HTTP i'm sending:
```http://192.168.0.124:80/json?plain=[{'data':[4594,2658,304,448,280,1050,304,1028,306,448,280,1052,306,448,306,446,306,448,304,448,302,1030,280,472,304,450,304,450,306,1026,306,448,306,446,304,448,308,1024,306,1026,304,450,280,474,302,450,306,446,304,448,306,448,304,448,304,448,304,450,302,450,280,472,280,474,304,448,306,450,302,448,304,448,306,448,282,1050,306,448,306,446,308,446,304,448,282,470,280,474,304,448,308,1024,306,446,308,446,282,472,302,450,306,448,304,1026,308,446,280,472,280,1050,304,448,306,446,306,448,304,448,304,450,280,1052,284,1046,306,448,280,472,306,430,304],'type':'raw','khz':38}]```
I'd like to note that when received the length is 66 but when sending it is 131
Anything else I can try, any ideas? Thanks and keep it up!

mdhiggins, 2018-05-03:
Lots of cheap AC units have trouble working due to the complexity and lack of standardization of the codes
Its unfortunately not an easy problem to fix
You can ask on the IRRemote github and see if they have any suggestions but its a limitation of the library due to the nature of the way they do their encoding

sorinGrecu, 2018-05-07:
I see. I will look further into it on the IRRemote github, thanks! Have a nice one!

closed 2018-05-07 by sorinGrecu