mdhiggins/ESP8266-HTTP-IR-Blaster #175: Device only stays joined to WiFi network for about 10 or 11 seconds

sdrasmussen, 2018-05-12:
Hi, all.  I am working my way through this project, and have hit a snag.  I can get the nodemcu to join my home WiFi network, and when it does, it answers ping and also web request.  However, it only stays joined to my WiFi network for about 10 or 11 seconds, after which it drops off, and will never join again, until I power cycle the device.  Is anyone else having this issue?

mdhiggins, 2018-05-12:
What version of lwIP are you using? and did you select the correct board? and what arduino ESP8266 version are you using?

Certain builds of ESP8266 and combinations with lwIP have memory leaks and cause networking problems

Also what hardware are you using?

sdrasmussen, 2018-05-12:
I am using 1.4 IwIP per your recommendation in the docs.  As for the arduino ESP8266 version, are you talking about my board?  If so, it is a nodemcu 1.0 with cp2102.  Not sure if I am answering your questions properly.  (extreme newb here)

mdhiggins, 2018-05-12:
Which build of the Arduino ESP8266 Core?
2.4.0 is the recommended one and later version have memory leaks

sdrasmussen, 2018-05-12:
2.4.1 is what I have installed.

mdhiggins, 2018-05-12:
Try rolling back to 2.4.0, 2.4.1 definitely is known to have a networking memory leak

Still make sure you use lwIP 1.4 prebuilt

From: sdrasmussen
Sent: Friday, May 11, 2018 10:48 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Device only stays joined toWiFi network for about 10 or 11 seconds (#175)

2.4.1 is what I have installed.

sdrasmussen, 2018-05-12:
That did it!  You are the best, man.  Thanks a million for your time.  I can't wait to get this thing finished.  I 3D printed a case for it...should turn out looking pretty decent.

mdhiggins, 2018-05-12:
Feel free to share!
Glad its working

closed 2018-05-12 by mdhiggins