mdhiggins/ESP8266-HTTP-IR-Blaster #176: Channel chaning

emogg, 2018-05-14:
Can you give more guidance on channel changing. I can control with channel up/down using
"http://xxx.xxx.xxx.xxx:667/msg?code=400401004C4D:PANASONIC:48&address=0x4004&pass=****", but I fail with channel numbering and naming.

My base url is "http://******.*****:667/json?pass=*****"
Button 1, for example is "data:'01000809',type:'PANASONIC',length:32,address=16388"
A channel named "bbc" and numbered "101"

I've tried "Alexa, change channel to 200 on TV" and get "Sorry TV appears to be malfunctioning"

Similar result with "Alexa, change channel to bbc on device"

Any hints?

mdhiggins, 2018-05-14:
It uses standard JSON formatting for the channels which you almost nailed except for the `=`

Should be:
`data:'01000809',type:'PANASONIC',length:32,address:16388`

emogg, 2018-05-14:
Thanks for catching that so quickly, but the results are just the same. i.e "TV malfunctioning". 
I cannot use the name "bbc" either.

From the json object:      "c1": "data:'01000809',type:'PANASONIC',length:32,address:16388",

Question 1: Where is the malfunctioning taking place? I'm using the phrasing from the alexa ChannelController web page.

Question 2: Do I say 101 as "one zero one", or "one hundred and one", possibly "one oh one"?

mdhiggins, 2018-05-14:
Email me your JSON discovery packet at the bottom of the page

emogg, 2018-05-17:
I've emailed my json discovery packet to you. Any progress? Is there anything I can pursue?

emogg, 2018-05-17:
Dunno how it got closed. Finger trouble I suppose

mdhiggins, 2018-05-17:
Don’t believe I ever got an email. Confirm which email you sent to please and try sending again

emogg, 2018-05-17:
Seems it went to "notifications@github.com". I just hit reply in thunderbird.
However here is the json contents:

[
    {
        "cookie": {
            "powerOn": "http://123.456.789.111:667/msg?code=40040100BCBD:PANASONIC:48&address=16388&repeat=15&rdelay=1&pass=abcdefgh",
            "powerOff": "http://123.456.789.111:667/msg?code=40040100BCBD:PANASONIC:48&address=16388&pass=abcdefgh",
            "volumeUp": "http://123.456.789.111:667/msg?code=400401000405:PANASONIC:48&address=0x4004&pass=abcdefgh",
            "volumeDown": "http://123.456.789.111:667/msg?code=400401008485:PANASONIC:48&address=0x4004&pass=abcdefgh",
            "mute": "http://123.456.789.111:667/msg?code=400401004C4D:PANASONIC:48&address=0x4004&pass=abcdefgh",
            "baseURL": "http://*********.local:667/json?pass=abcdefgh",
            "cup": "http://123.456.789.111:667/msg?code=400401004C4D:PANASONIC:48&address=0x4004&pass=abcdefgh",
            "cdown": "http://123.456.789.111:667/msg?code=40040100ACAD:PANASONIC:48&address=0x4004&pass=abcdefgh",
            "c1": "data:'01000809',type:'PANASONIC',length:32,address:16388",
            "c2": "data:'01008889',type:'PANASONIC',length:32,address:16388",
            "c3": "data:'01004849',type:'PANASONIC',length:32,address:16388",
            "c4": "data:'0100C8C9',type:'PANASONIC',length:32,address:16388",
            "c5": "data:'01002829',type:'PANASONIC',length:32,address:16388",
            "c6": "data:'0100A8A9',type:'PANASONIC',length:32,address:16388",
            "c7": "data:'01006869',type:'PANASONIC',length:32,address:16388",
            "c8": "data:'0100E8E9',type:'PANASONIC',length:32,address:16388",
            "c9": "data:'1000809',type:'PANASONIC',length:32,address:16388",
            "c0": "data:'01009899',type:'PANASONIC',length:32,address:16388",
            "cdot": "data:'01001819',type:'PANASONIC',length:32,address:16388",
            "cent": "data:'01009293',type:'PANASONIC',length:32,address:16388"
        },
        "friendlyName": "TV",
        "endpointId": "living-room-tv",
        "description": "Living Room TV via IR Controller",
        "manufacturerName": "MD Higgins",
        "displayCategories": [
            "TV"
        ],
        "capabilities": [
            {
                "interface": "Alexa.PowerController",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.StepSpeaker",
                "version": "1.0",
                "type": "AlexaInterface"
            },
            {
                "interface": "Alexa.ChannelController",
                "version": "1.0",
                "type": "AlexaInterface",
                "properties": {
                    "supported": [
                        {
                            "name": "channel"
                        }
                    ]
                }
            }
        ]
    }
]

emogg, 2018-05-17:
Well, not really. The IP and pw were dummies. perhaps I should have used more asterisks. Where should I send an email to instead?

mdhiggins, 2018-05-17:
That's ok, I found your actual database entry so I could see the unedited version

Looks like you left your .local IP address in the base URL instead of your external IP address unlike all the other URLs

emogg, 2018-05-18:
AH, OK. These old eyes are a bit dim! Thank you for the heads up. It is now possible to change channels. The problem now is to find the most efficient way of saying the channel names for repeatability.

emogg, 2018-05-18:
Oh, I only meant whether in the IR Controller Device Setup it is better to use "bbc1", "bbc one", or "b.b.c.1" for channel naming etc. I'd like to know what others use. Anyway, for me, this issue is closed. Thanks for all your help.

closed 2018-05-18 by emogg

mdhiggins, 2018-05-19:
I actually wrote a long algorithm that will try and check against most possibilities that you listed above so that a match can be found regardless but the safest bet is the simplest bbc1 option, that being said I think most of those should be recognized