mdhiggins/ESP8266-HTTP-IR-Blaster #178: Absolute beginer here

diydude, 2018-05-19:
It took me hours for the ide to let me upload the file to the esp8266, when it finished uploading I tried configuring it but it didn’t showed up in the wifi list, what could be the issue? I’ve probably done something wrong. 
It says “config gpio pin”

diydude, 2018-05-20:
I tried on two esp8266 boards and the issue is the same or did you mean my computer?

mdhiggins, 2018-05-20:
I meant the boards but if 2 boards are doing it then you need to verify your Arduino IDE settings
What boards are you using specifically and what upload settings are you using?

diydude, 2018-05-20:
I've used this one https://www.amazon.com/gp/product/B01IK9GEQG/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1 it is the one on the shopping list, I think it would be easier if you tell me the settings I need to use.

mdhiggins, 2018-05-21:
That’s the board I use and did all the development on so it should be fine with all the default settings but its definitely not what’s pictured in the images you sent me

The Amazon link is a NodeMCU V2, the one you’re using is the unofficial NodeMCU V3 by LoLin

Did you try any of the things I put in bold in that long email?

From: diydude
Sent: Sunday, May 20, 2018 6:34 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Absolute beginer here (#178)

I've used this one https://www.amazon.com/gp/product/B01IK9GEQG/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1 it is the one on the shopping list, I think it would be easier if you tell me the settings I need to use.

mdhiggins, 2018-05-21:
Whoops mixed up the people who've been messaging me today

diydude, 2018-05-21:
I tried with another usb cable just now and it doesn't work either, this are the screen shots
[image]
this is what it shows on the log

[image]
and this are the settings

mdhiggins, 2018-05-24:
Alright your settings are all messed up
Change board to NodeMCU 1.0 (ESP-12E Module)
Change Flash Size to 4M (3M SPIFFS)
Change lwIP to 1.4 Prebuilt
Keep 115200
Change CPU frequency to 80 Mhz

Give that a shot

And make sure you’re on 2.4.0 for ESP8266 Arduino Core

diydude, 2018-05-25:
[image]

this is what i got now.

mdhiggins, 2018-05-25:
Well that's significant progress! It booted and probably has an accessible webpage
External stuff didn't work but that usually means either you didn't use 2.4.0 and lwIP 1.4 or you configured your networking settings incorrectly

mdhiggins, 2018-05-25:
In your Arduino IDE go to:
Tools > Board > Board Manager
Type ESP8266
Click on the one option that should appear
Click on select version
Change to 2.4.0 and hit update

After that's done installing go back to lwIP and v1.4 prebuilt should be there now

diydude, 2018-05-25:
the IR code on my AC does not work, it sends it but it does nothing but I've tried it on my tv and it does, other than that there is nothing more to say than thanks!

mdhiggins, 2018-05-25:
Very common for AC not to work; often a ton of weird complications with AC units for whatever reason. You can check on https://github.com/markszabo/IRremoteESP8266 and see if someone has posted about your model

closed 2018-05-25 by mdhiggins