mdhiggins/ESP8266-HTTP-IR-Blaster #181: IR Blasters going offline

OzzieJacks, 2018-05-25:
I have two IR Blasters and at various times during the day Alexa will say there is no device by the name 'device name'.  If I check on the alexa admin web page it shows them as offline.  If I perform a discovery they will come back online and work again.  They will also start working on there own if I do nothing and just wait a few hours.

When I try to connect to them from outside my LAN they always seem to be accessible and working correctly.  Both devices always go offline at the same time which would indicate some kind of connection problem but I can't detect anything wrong.

What is the mechanism that amazon uses to check if devices are online?  What steps can I take to debug what is happening?

mdhiggins, 2018-05-25:
Alexa actually just pings the skill to make sure the skill is online, so aside from pythonanywhere doing maintenance there is no periodic device ping and even if the device is offline that doesn't actually result in any behavior change; alexa will still attempt to issue the commands. No device by device name is probably more of a name recognition issue than anything else did. Did you check to see how Alexa interpreted your verbal command in the history to see if it heard the wrong device name?

OzzieJacks, 2018-05-26:
The devices just stopped responding again. These are logs from my history

alexa turn on stereo
Today at 7:03 PM on Alpha
alexa turn on tv
Today at 7:10 PM on Alpha
alexa change channel to c. n. n. on cable
Today at 7:16 PM on Alpha

Alexa would respond with.
"I couldn't find a device or group named TV in Michael's profile."
Same for stereo but the change channel commands on cable surprisingly work.

The device list shows:
Stereo (Offline)
Bedroom Stereo via IR Controller
TV (Offline)
Living Room TV via IR Controller
Cable (Offline)
Living Room Tivo via IR Controller

I just pressed discover devices and then get:
Stereo
Bedroom Stereo via IR Controller
TV
Living Room TV via IR Controller
Cable
Living Room Tivo via IR Controller

The same commands that didn't work start instantly working.  I can try renaming TV and Stereo to something else and see if they continue to work.  What device names have you found to work?  Strange that the discovery action would cause the devices to work if amazon had a problem with the names.

mdhiggins, 2018-05-26:
And did you check your Alexa history to see how those commands were interpreted yet?

From: OzzieJacks
Sent: Friday, May 25, 2018 10:40 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] IR Blasters going offline(#181)

The devices just stopped responding again. These are logs from my history
alexa turn on stereo
Today at 7:03 PM on Alpha
alexa turn on tv
Today at 7:10 PM on Alpha
alexa change channel to c. n. n. on cable
Today at 7:16 PM on Alpha
Alexa would respond with.
"I couldn't find a device or group named TV in Michael's profile."
Same for stereo but the change channel commands on cable surprisingly work.
The device list shows:
Stereo (Offline)
Bedroom Stereo via IR Controller
TV (Offline)
Living Room TV via IR Controller
Cable (Offline)
Living Room Tivo via IR Controller
I just pressed discover devices and then get:
Stereo
Bedroom Stereo via IR Controller
TV
Living Room TV via IR Controller
Cable
Living Room Tivo via IR Controller
The same commands that didn't work start instantly working. I can try renaming TV and Stereo to something else and see if they continue to work. What device names have you found to work? Strange that the discovery action would cause the devices to work if amazon had a problem with the names.

OzzieJacks, 2018-05-26:
Yes here is a copy paste from settings->history to show how Alexa interpreted the commands.
Before discovery.
Today at 7:10 PM
alexa turn on tv 
> I couldn't find a device or group named TV in Michael's profile.

After discovery.
Today at 7:33 PM
alexa turn on tv 
>ok

Same command and interpretation but different response.

mdhiggins, 2018-05-26:
I would try not using the generic name TV as Amazon uses this for smart redirection to local/grouped devices and may be causing conflicts; there appear to be people on the developer forums that have encountered this error

If you name something for example “bedroom tv” and set the device type to TV and then group it with the appropriate echo is will still respond to “turn on TV”

From: OzzieJacks
Sent: Friday, May 25, 2018 10:56 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] IR Blasters going offline(#181)

Yes here is a copy paste from settings->history to show how Alexa interpreted the commands.
Before discovery.
Today at 7:10 PM
alexa turn on tv
I couldn't find a device or group named TV in Michael's profile.
After discovery.
Today at 7:33 PM
alexa turn on tv
ok
Same command and interpretation but different response.

OzzieJacks, 2018-05-26:
I have renamed the devices.  I will report back in a few days to let you know how it goes.  Thanks for your help.

mdhiggins, 2018-06-08:
Glad its working!

closed 2018-06-08 by mdhiggins